google-coral/coralmicro #22: Coral Micro examples on the Arduino Legacy IDE (1.8.X)

hpssjellis, 2023-02-11:
### Description

Is there any plan to allow the Coral-micro examples to be used on the older [Legacy IDE (1.8.X)](https://www.arduino.cc/en/software) ?

As the new Arduino IDE 2.x had some hiccups when it was first introduced, some educational facilities were slow to update to the new IDE, is there a plan to allow Coral Micro to load the examples for the old IDE?

P.S. The board installs fine on the old IDE, I have not tested it as I don't have a Coral-micro yet.

### Issue Type

Build/Install

### Operating System

Windows 10, Ubuntu

hjonnala, 2023-02-13:
Unfortunately, No. We don't have any plans to support old IDE. https://coral.ai/docs/dev-board-micro/arduino/#1-gather-requirements

[image]

hpssjellis, 2023-02-13:
Thanks @hjonnala for the reply. I can close this issue. I have a few other questions but will put them in their own issues.

closed 2023-02-13 by hpssjellis