google-coral/coralmicro #25: X and Y coordinates are replaced by score values in posenet_decoder output

Ramsonjehu, 2023-02-21:
### Description

I'm running [posenet cpu model](https://raw.githubusercontent.com/google-coral/test_data/master/posenet/posenet_mobilenet_v1_075_353_481_16_quant_decoder.tflite). The model has detected 4 poses which is correct by the way. but the x and y co-ordinate values are replaced by the score values for the first two poses as you can see below the from output of FormatPosenetOutput function. 
*************************************************************************************
Num Poses: 4

Pose 0 -- score: 0.955521

Keypoint NOSE -- x=0.991659,y=0.643709,score=0.995698

Keypoint LEFT_EYE -- x=0.991455,y=0.992649,score=0.992551

Keypoint RIGHT_EYE -- x=0.988241,y=0.982647,score=0.975114

Keypoint LEFT_EAR -- x=0.980253,y=0.964800,score=0.942884

Keypoint RIGHT_EAR -- x=0.982077,y=0.965587,score=0.643709

Keypoint LEFT_SHOULDER -- x=0.950993,y=0.981304,score=0.991659

Keypoint RIGHT_SHOULDER -- x=0.993735,y=0.922237,score=0.992649

Keypoint LEFT_ELBOW -- x=0.993116,y=0.954117,score=0.991455

Keypoint RIGHT_ELBOW -- x=0.958017,y=0.614445,score=0.982647

Keypoint LEFT_WRIST -- x=0.996635,y=0.993426,score=0.988241

Keypoint RIGHT_WRIST -- x=0.907832,y=0.992296,score=0.964800

Keypoint LEFT_HIP -- x=0.820433,y=0.992142,score=0.980253

Keypoint RIGHT_HIP -- x=0.987976,y=0.988401,score=0.965587

Keypoint LEFT_KNEE -- x=0.963107,y=0.982735,score=0.982077

Keypoint RIGHT_KNEE -- x=0.888850,y=0.893946,score=0.981304

Keypoint LEFT_ANKLE -- x=0.996410,y=0.997877,score=0.950993

Keypoint RIGHT_ANKLE -- x=0.990097,y=0.988523,score=0.922237

Pose 1 -- score: 0.936542

Keypoint NOSE -- x=0.996655,y=0.422413,score=0.993735

Keypoint LEFT_EYE -- x=0.990343,y=0.995881,score=0.954117

Keypoint RIGHT_EYE -- x=0.951878,y=0.973267,score=0.993116

Keypoint LEFT_EAR -- x=0.970227,y=0.970716,score=0.614445

Keypoint RIGHT_EAR -- x=0.949837,y=0.969003,score=0.958017

Keypoint LEFT_SHOULDER -- x=0.801735,y=0.816434,score=0.993426

Keypoint RIGHT_SHOULDER -- x=0.995525,y=0.951153,score=0.996635

Keypoint LEFT_ELBOW -- x=0.961644,y=0.983706,score=0.992296

Keypoint RIGHT_ELBOW -- x=0.672398,y=0.941005,score=0.907832

Keypoint LEFT_WRIST -- x=0.994890,y=0.992121,score=0.992142

Keypoint RIGHT_WRIST -- x=0.821764,y=0.992328,score=0.820433

Keypoint LEFT_HIP -- x=0.801571,y=0.978170,score=0.988401

Keypoint RIGHT_HIP -- x=0.981952,y=0.971370,score=0.987976

Keypoint LEFT_KNEE -- x=0.958127,y=0.940331,score=0.982735

Keypoint RIGHT_KNEE -- x=0.812278,y=0.915007,score=0.963107

Keypoint LEFT_ANKLE -- x=45.939163,y=306.190369,score=0.893946

Keypoint RIGHT_ANKLE -- x=54.945610,y=309.451660,score=0.888850

Pose 2 -- score: 0.925438

Keypoint NOSE -- x=226.782761,y=183.224136,score=0.997877

Keypoint LEFT_EYE -- x=228.083115,y=173.909515,score=0.996410

Keypoint RIGHT_EYE -- x=217.703690,y=174.543243,score=0.988523

Keypoint LEFT_EAR -- x=239.427628,y=176.000122,score=0.990097

Keypoint RIGHT_EAR -- x=212.744202,y=177.428848,score=0.422413

Keypoint LEFT_SHOULDER -- x=256.687012,y=203.382462,score=0.996655

Keypoint RIGHT_SHOULDER -- x=210.265442,y=207.248230,score=0.995881

Keypoint LEFT_ELBOW -- x=269.823792,y=245.704254,score=0.990343

Keypoint RIGHT_ELBOW -- x=203.039642,y=247.923218,score=0.973267

Keypoint LEFT_WRIST -- x=245.640976,y=267.407898,score=0.951878

Keypoint RIGHT_WRIST -- x=224.644806,y=265.901062,score=0.970716

Keypoint LEFT_HIP -- x=258.826752,y=268.022217,score=0.970227

Keypoint RIGHT_HIP -- x=218.966782,y=263.892670,score=0.969003

Keypoint LEFT_KNEE -- x=269.871460,y=277.554321,score=0.949837

Keypoint RIGHT_KNEE -- x=199.246002,y=265.904022,score=0.816434

Keypoint LEFT_ANKLE -- x=264.594391,y=316.269043,score=0.801735

Keypoint RIGHT_ANKLE -- x=219.024536,y=300.613800,score=0.951153

Pose 3 -- score: 0.924364

Keypoint NOSE -- x=152.958649,y=51.454514,score=0.995525

Keypoint LEFT_EYE -- x=158.717789,y=47.946064,score=0.983706

Keypoint RIGHT_EYE -- x=149.958878,y=46.098427,score=0.961644

Keypoint LEFT_EAR -- x=168.336899,y=53.966148,score=0.941005

Keypoint RIGHT_EAR -- x=145.987793,y=52.476761,score=0.672398

Keypoint LEFT_SHOULDER -- x=175.504486,y=89.579712,score=0.992121

Keypoint RIGHT_SHOULDER -- x=132.421005,y=78.113968,score=0.994890

Keypoint LEFT_ELBOW -- x=182.233734,y=128.720718,score=0.992328

Keypoint RIGHT_ELBOW -- x=115.090141,y=121.977531,score=0.821764

Keypoint LEFT_WRIST -- x=179.247437,y=163.048798,score=0.978170

Keypoint RIGHT_WRIST -- x=109.832756,y=160.000000,score=0.801571

Keypoint LEFT_HIP -- x=164.019348,y=171.155273,score=0.971370

Keypoint RIGHT_HIP -- x=134.367416,y=170.979340,score=0.981952

Keypoint LEFT_KNEE -- x=141.328262,y=239.238464,score=0.940331

Keypoint RIGHT_KNEE -- x=133.383041,y=237.994507,score=0.958127

Keypoint LEFT_ANKLE -- x=159.236404,y=304.678497,score=0.915007

Keypoint RIGHT_ANKLE -- x=120.150841,y=296.735199,score=0.812278
******************************************************************************************
Seems like some kind of memory over-write problem in the posenet_decoder implementation.  Please help solving this issue. 

Thanks & Regards,
Ramson Jehu K

### Issue Type

Bug

### Operating System

Ubuntu

### Coral Device

Dev Board

### Programming Language

C++

hjonnala, 2023-02-21:
Could you share the script that you are trying to run here please.Thanks!

Ramsonjehu, 2023-02-22:
I'm not using any script, Indeed what i did is to evaluate the model. I have added the posenet_decoder op implementation from here (https://github.com/google-coral/coralmicro/blob/main/libs/tensorflow/posenet_decoder_op.cc) to the tflite micro library and ran it locally on the PC. Do you need the entire source?

hjonnala, 2023-02-22:
Yes, please share all the reproducable steps, files etc.  

- And are you planning to run this model on any microcontrollers? 
- Have you tested any other [lower resolution models](https://coral.ai/models/pose-estimation/#trained-models)?

Ramsonjehu, 2023-02-24:
I have uploaded the source files in google drive, Here is the [link](https://drive.google.com/file/d/1sSXpEm1RJGPY-q9riSTjAEE-qd97n75E/view?usp=sharing). Please find the build and execution steps in the readme.txt file. 

To answer your questions,
And are you planning to run this model on any microcontrollers?
Yes, I'm planning to run on NXP RT1170 microcontroller.

Have you tested any other [lower resolution models](https://coral.ai/models/pose-estimation/#trained-models)?
I think this was the least resolution model in Posenet. which was 353x481x3.

hjonnala, 2023-02-27:
Thanks for sharing the details.. wondering do you have any CPU tflite model with same width and heght?

Ramsonjehu, 2023-02-28:
I used the PoseNet MobileNet V1 CPU model in that source which was taken from [https://coral.ai/models/pose-estimation/](https://coral.ai/models/pose-estimation/).

hjonnala, 2023-02-28:
I used [detect_poses](https://github.com/google-coral/coralmicro/blob/main/examples/detect_poses/detect_poses.cc#L44) script with  your test image to test 324x324 and 353x481 edgetpu models on Dev Board Micro.

The bin file got generated with the below snippet:

```
python3 -c "import sys,PIL.Image;sys.stdout.buffer.write(PIL.Image.open(sys.stdin.buffer).resize((481, 353)).convert('RGB').tobytes())" <input_image_rgb.jpg >input_i
mage_rgb_481x353.bin
```
And it can visulaized with the below snippet:
```
python3 -c "import sys,PIL.Image;PIL.Image.frombytes('RGB', (481, 353), sys.stdin.buffer.read()).show()" <input_image_rgb_481x353.bin 
```

The results look fine with [324x324](https://github.com/hjonnala/snippets/blob/main/test_log.txt#L2) edgetpu model and I am  getting similar results as yours with [353x481x3](https://github.com/hjonnala/snippets/blob/main/test_log.txt#L81) edgeptu model. Unfortuantley, I don't have 324x324 cpu model to share with you.

I would suggest to generate a 324x324 cpu model to run on NXP RT1170 microcontroller. And test 353x481 model with other images having people(2,3,4,5 etc..) to identify any other limitations.
Also, Feel free to inspect values directly from [output tensors](https://github.com/google-coral/coralmicro/blob/1dde7900d5d2d57efdf2f5a2a969ee72eb68946b/libs/tensorflow/posenet.cc#L44). Thanks!

Ramsonjehu, 2023-03-02:
Hi @hjonnala,
I think this issue is with the posenet_decoder implementation in [posenet_decoder_op.cc](https://github.com/google-coral/coralmicro/blob/main/libs/tensorflow/posenet_decoder_op.cc). 
why because, the same model when run using TFLite interpreter. The output is correct. Please checkout this [colab notebook](https://colab.research.google.com/drive/15I2_gAmgmFT18wRm3mJu04H6uXrN78oO?usp=sharing). Upload this [zip file](https://drive.google.com/file/d/1q7mV13NyEmCrQVMOu2UuGq0_USWYN4D8/view?usp=sharing) to the colab for running the code. 

> I would suggest to generate a 324x324 cpu model

Can you please let me know the steps for generating 324x324 cpu model?

Ramsonjehu, 2023-03-06:
hi @hjonnala ,
Any updates on this?

Thanks

hjonnala, 2023-03-06:
Hi, I have reported this issue to the internal team. Unfortuantley, as of now they don't have any bandwidth to look into this..Thanks!