google-coral/coralmicro #26: USB passthrough

EnariSaarBolt, 2023-03-09:
### Description

Is there a way to make RT1176 passthrough usb directly to coral accelerator, since usb acceleration is out of stock.

### Issue Type

Others

### Operating System

Linux

### Coral Device

Dev Board

### Programming Language

C++

hjonnala, 2023-03-10:
I think, you are referring to [i.MX RT1176 uCOM module](https://www.cnx-software.com/2021/02/26/embedded-artists-launches-1ghz-nxp-i-mx-rt1176-crossover-mcu-module-and-devkit/). We don't have any devices other than the [Coral USB accelrator module](https://coral.ai/products/accelerator) to directly connected to host machines via USB. If possible, feel free to [prototype](https://tony-kim.medium.com/prototyping-with-the-google-coral-ml-tpu-accelerator-module-usb-2-0-high-speed-f0e19218cec2) and integrate the [Coral Accelerator Module](https://coral.ai/products/accelerator-module) into your custom PCB hardware design.

EnariSaarBolt, 2023-03-10:
I mean this https://coral.ai/products/dev-board-micro, is there possibility to turn this into usb accelerator without buying one specially made for it.

hjonnala, 2023-03-10:
No, its not possible.

closed 2023-03-10 by hjonnala