google-coral/coralmicro #28: Camera images have color patches

metanav, 2023-03-16:
I am running default multicore_model_cascade on the micro dev board and streaming the images over usb on a Linux virtualbox.

```
$ python3 scripts/flashtool.py -a multicore_model_cascade

$ python3 apps/multicore_model_cascade/multicore_model_cascade.py

```
The person/pose detection works but the images have always some kind of round color patches as you can see in the screenshot below.

[image]

Is it normal or my micro dev camera have some defects?

hjonnala, 2023-03-17:
> Is it normal or my micro dev camera have any lens/sensor defects? Since I plan to use the RGB images for a custom model and my use case depends on the color of the objects so please confirm it.

yes, it is normal.

hjonnala, 2023-03-17:
I think, it depends on camera angle and exposure to the light. Please try placing dev board micro in different positions.

metanav, 2023-03-18:
I have tried all angles and positions in different lighting conditions (day/night, indoor/outdoor) but the image always have spots.

josecols, 2023-03-27:
I noticed that part of the color flare is caused by the camera and TPU LEDs, which is especially true if you use the case. A workaround that helped me is reducing the brightness of the TPU LED with `LedSetBrightness(Led::kTpu, 1);`. But unfortunately, the API does not exposes the camera LED.

Another option is using the [DiscardFrames function](https://coral.ai/docs/reference/micro/camera/#_CPPv4N10coralmicro10CameraTask13DiscardFramesEi) to auto-calibrate the sensor.

metanav, 2023-03-27:
I tried to stick a black tape over the camera and TPU LED which seems not helpful. I will try to use the [DiscardFrames function](https://coral.ai/docs/reference/micro/camera/#_CPPv4N10coralmicro10CameraTask13DiscardFramesEi) later. By the way, do you see the similar color flares or less as I have on the images in the previous posts?

hjonnala, 2023-04-03:
Hi @metanav unfortunately, color flares are expected with the on board camera. Please test your custom model on dev board micro. If the performance is not as expected, please try  [Dev Board](https://coral.ai/products/dev-board) with a USB camera or if possible, please design a custom board with [accelerator-module](https://coral.ai/products/accelerator-module) as per your requirements.

Thanks!!

closed 2023-04-26 by hjonnala