google-coral/coralmicro #29: Failing to upload via Arduino IDE in Windows 10

Hanqaqa, 2023-03-17:
### Description

I followed the instructions here:
https://coral.ai/docs/dev-board-micro/arduino/#2-install-the-coral-package
Installed the additional boards and tried uploading the BlinkLed Basics Example. And I keep getting this `No module named 'progress'`.
I tried reinstalling via pip the progress package and it still isn't working. 
I tried in a second computer (Windows 10) and I keep getting the same error.

Windows 10 isn't mentioned anywhere in the documentation, So my question is:
is there any way of uploading code using the Arduino IDE in Windows 10? Or am I forced to switch into a Linux partition?

### Issue Type

Build/Install

### Operating System

Windows 10

### Coral Device

Dev Board

### Programming Language

Other

### Relevant Log Output

```shell
[5620] Failed to execute script 'flashtool' due to unhandled exception!
Traceback (most recent call last):
  File "flashtool.py", line 16, in <module>
ModuleNotFoundError: No module named 'progress'
Failed uploading: uploading error: exit status 1
```

hjonnala, 2023-03-17:
Hi @Hanqaqa we don't have any support ardunio IDE on windows. However please try set up the [environment](https://github.com/google-coral/coralmicro/blob/main/.github/workflows/ci.yml#L148) and build it locally on GIT bash using the command at: https://github.com/google-coral/coralmicro/blob/main/.github/workflows/ci.yml#L156.. Thanks!!

Hanqaqa, 2023-03-20:
Thanks hjonnala.

Unfortunately in Windows 10 I run into errors after I have cloned coralmicro and all submodules with the git bash. When I bash build with the following arguments as you said
`bash build.sh -n -a -g -s`
I get the following errors in the git console 

```
$ bash build.sh -n -a -g -s
+ cmake -B /c/Users/MarcoVM/Desktop/coralmicro/build -G Ninja -S /c/Users/MarcoVM/Desktop/coralmicro
-- Fetching https://developer.arm.com/-/media/Files/downloads/gnu-rm/9-2020q2/gcc-arm-none-eabi-9-2020-q2-update-win32.zip
-- Toolchain prefix: C:/Users/MarcoVM/Desktop/coralmicro/third_party/toolchain-win/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi
-- Toolchain prefix: C:/Users/MarcoVM/Desktop/coralmicro/third_party/toolchain-win/gcc-arm-none-eabi-9-2020-q2-update/lib/gcc/arm-none-eabi
-- The C compiler identification is GNU 9.3.1
-- The CXX compiler identification is GNU 9.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/MarcoVM/Desktop/coralmicro/third_party/toolchain-win/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/MarcoVM/Desktop/coralmicro/third_party/toolchain-win/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/MarcoVM/Desktop/coralmicro/third_party/toolchain-win/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc.exe
-- Configuring done (26.0s)
CMake Error at cmake/toolchain-arm-none-eabi-gcc.cmake:230 (add_library):
  Cannot find source file:

    C:/Users/MarcoVM/Desktop/coralmicro/third_party/nxp/rt1176-sdk/components/osa/fsl_os_abstraction_free_rtos.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc
Call Stack (most recent call first):
  libs/nxp/rt1176-sdk/CMakeLists.txt:231 (add_library_m7)
```
Then, the same CMake errors keeps getting repeated for 30 or 40 times with different toolchains
I have tried in two Windows 10 computers, making sure I have installed chocolatey in both of them, as well as cmake and ninja and installed the Python requirements.txt. But I still get the same errors
After getting tired I decided to just boot up my external SSD with a Ubuntu, build it from source and everything worked fine.
If you see anything I might have missed in Windows 10 please tell me. If not this issue can be closed

hjonnala, 2023-03-20:
I am able to falsh, blibk_led example, with Arduino IDE on windows 11 with some changes ( Haven't tested for other examples though). Please try the below steps and let me know if it works for you as well. Thanks!!

1. FInd the flashtool.exe and replace with this [flashtool](https://github.com/hjonnala/snippets/blob/main/Dev_Board_Micro/win/flashtool.exe). On my system it is located at: `C:\Users\heman\AppData\Local\Arduino15\packages\coral\tools\flashtool\1.0.0`
2. git clone --recurse-submodules -j8 https://github.com/google-coral/coralmicro
3. copy coralmicro/third_party to root_dir. On my system root_dir is `C:\Users\heman\AppData\Local\Temp `

Hanqaqa, 2023-03-20:
@hjonnala Thank you so much.

Copying your flashtool.exe (Windows 10 alerted me it was a virus, I had to manually allow that file to exist) and copying the third party folder worked great in my Windows 10. 
I have flashed a few examples in the Arduino 2 IDE, incluiding some Tensorflow examples and they have all worked perfectly.
My question is, would it be hard to change the source files here so it was easier to install on Arduino IDE 2 on Windows 10?

Now it has just been copying and pasting two files. Would it be too hard to make it Windows 10 compatible? I have already seen some other people complaining they couldn't get the Coral Micro running on Windows.

atvrager, 2023-03-20:
@Hanqaqa Regarding your issue building via build.sh -- It's likely that Git didn't check out all the files due to path length issues. In our CI runners, we enable longpaths support: https://github.com/google-coral/coralmicro/blob/main/.github/workflows/ci.yml#L136

Alternately, you could clone the repository to something with a shorter prefix (e.g. C:\coralmicro).

I've reproduced the issue with flashtool not working in Arduino -- seems that PyInstaller didn't bring in all the dependencies -- will take a look into a patch and new release to fix this, but hopefully for the time being @hjonnala's solution has you unblocked.

hjonnala, 2023-03-20:
Closing as the issue got fixed with this new release: https://github.com/google-coral/coralmicro/releases/tag/arduino-1.0.1

closed 2023-03-20 by hjonnala