google-coral/coralmicro #33: Issue accessing Jtag on Coral Micro Dev Board

Tarek-Mhareb, 2023-03-21:
### Description

Hello,

We are unable to connect using JTAG on the board.
The pictures shows the simple setup but the connection fails every time.
We tried Keil debuggers and we are still unable to communicate with the board.
Do we have to use an external power supply and free the USB port?

Thanks

[image]
[image]
[image]
[image]
[image]

### Issue Type

Build/Install

### Operating System

Windows 10, Ubuntu

### Coral Device

Dev Board Micro

hjonnala, 2023-03-21:
Hi @Tarek-Mhareb can you please try the set up software section mentioned at: https://coral.ai/docs/dev-board-micro/wireless-addon/#jtag

hjonnala, 2023-03-28:
Hi @Tarek-Mhareb were you able to access the jtag?

Tarek-Mhareb, 2023-03-29:
Hello @hjonnala 

I am using Linux Ubuntu and SWD (Jtag) now.
I am getting the following error every time I try to connect.
[Jlink CMD Log](https://github.com/google-coral/coralmicro/files/11101650/Jlink.CMD.Log)

Thanks,
Tarek

Tarek-Mhareb, 2023-03-29:
Hi @hjonnala 
I forgot to mention that it is attached on a custom board. It is not possible to connect to all jtag pins from interface.
[image]

hjonnala, 2023-03-29:
Does the custom board has  board to board connectors? https://coral.ai/docs/dev-board-micro/get-started/#2-get-to-know-the-board. I don't think, it would work without those connections.

Tarek-Mhareb, 2023-03-31:
Hi,
I also attached here the command line steps with the error.

[image]

hjonnala, 2023-04-03:
Hi, It looks like you are trying to debug without all the JTAG pins, since they are only connecting to the headers and not all the JTAG pins are broken out there. Unfortunately, this isn't supported.

Here are the some recommendation for you:

1. You can buy the wifi board and use that for debug. This is the only officially supported JTAG debugging method. 

2. or you can use [altium design files](https://github.com/google-coral/electricals/tree/master/dev_board_micro) to locate the Dev Board Micro pads with the remaining JTAG signals and solder their own wires to the PCB to connect to their debugger. 

3. or you can design another board that does break out the all JTAG pins available on the WIFI board from the high density connectors.

Thanks!

Tarek-Mhareb, 2023-04-12:
Hello @hjonnala ,

I was able after several tries to use JTAG. I can use both modes, but it seems that the USB has to be also connected in serial mode. JTAG can also be verified using windows/Linux JTAG Commander.
You can mark this as solved but I have a question related to flashing bare metal projects.
I used Mcuxpresso with JTAG and I was able to debug but not flash. I linked the application to RAM and performed real time debugging. I was not able to flash to QSPI memory onboard because it seems that the memory starting address is not at 0x30000000. Could you please provide us with the correct configuration for flashing C projects?

Thank you

hjonnala, 2023-04-12:
> I was able after several tries to use JTAG. I can use both modes, but it seems that the USB has to be also connected in serial mode. JTAG can also be verified using windows/Linux JTAG Commander.

Great, Thanks for letting us know.

> I used Mcuxpresso with JTAG and I was able to debug but not flash. I linked the application to RAM and performed real time debugging. I was not able to flash to QSPI memory onboard because it seems [...]

Unfortunately, I don't have experience with Mcuxpresso IDE. Please use the setps mentioned for [FreeRTOS](https://coral.ai/docs/dev-board-micro/freertos/) or [arduino IDE](https://coral.ai/docs/dev-board-micro/arduino/) to flash the Dev Board Micro. Thanks!!

closed 2023-04-12 by hjonnala

wawanUnic, 2024-08-24:
@Tarek-Mhareb,
Hello. Tell me, will the following programmers be suitable for this microcontroller iMXRT1176?

1. https://www.waveshare.com/st-link-v2-mini-stm32.htm
2. https://www.waveshare.com/st-link-v2-mini-stm32.htm

Will they be able to program it? Will it be possible to debug it?