google-coral/coralmicro #37: How can I connect a coral dev board micro to a coral dev board?

dylan-leslie, 2023-05-02:
### Description

**What I am trying to achieve**
I would like to use a coral dev board as a host which can run some modified python example scripts as a client and receive image data from three USB connected coral dev board microcontrollers and store image data on the client.

**Some context**

For example, I would like to flash three coral dev board microcontrollers with the camera_streaming.cc example and give them each a unique IP address (192.168.101.10, 192.168.102.10 and 192.168.103.10). Like so:
```
python3 scripts/flashtool.py -e camera_streaming_rpc --subapp camera_streaming_rpc_usb --usb_ip_address 192.168.101.10
python3 scripts/flashtool.py -e camera_streaming_rpc --subapp camera_streaming_rpc_usb --usb_ip_address 192.168.102.10
python3 scripts/flashtool.py -e camera_streaming_rpc --subapp camera_streaming_rpc_usb --usb_ip_address 192.168.103.10
```

Then run a python script on a coral dev board which acts as a client to request images from each coral microcontroller.

I have the above functionality working when connecting the three microcontrollers to my ubuntu 20 machine (I can run a python client that makes GET requests to each coral microcontroller and store the response data).

**My main problem**
When connecting a coral dev board microcontroller to a coral dev board I am unable to see any devices on usb0 or usb1 the same way I can on an ubuntu 20 machine. I tried to follow the instructions [here](https://github.com/google-coral/edgetpu/issues/672) by configuring the `/etc/NetworkManager/system-connections/gadget0.nmconnection` and `/etc/NetworkManager/system-connections/gadget1.nmconnection` files to the same domains as the USB ip address that I flashed. E.g
```
mendel@zippy-tang:~$ sudo cat /etc/NetworkManager/system-connections/gadget1.nmconnection 
[connection]
id=gadget1
uuid=6c85437c-ba59-4522-baf0-bf68e006b601
type=ethernet
interface-name=usb1
permissions=

[ethernet]
mac-address-blacklist=

[ipv4]
address1=192.168.103.2/24
dns-search=
method=manual

[ipv6]
addr-gen-mode=stable-privacy
dns-search=
method=auto
```
However the above has no success as I still cannot ping the board and receive a response.
```
mendel@zippy-tang:~$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 28:bd:89:bc:a0:96 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.70/24 brd 192.168.178.255 scope global dynamic noprefixroute eth0
       valid_lft 7120sec preferred_lft 7120sec
    inet6 fe80::797c:2a5c:c17b:6afa/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 3000
    link/ether 28:bd:89:bc:a0:97 brd ff:ff:ff:ff:ff:ff
    inet 192.168.178.141/24 brd 192.168.178.255 scope global dynamic noprefixroute wlan0
       valid_lft 7120sec preferred_lft 7120sec
    inet6 fe80::3f3d:7c3e:c48c:a54f/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
4: p2p0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 3000
    link/ether 00:0a:f5:89:89:81 brd ff:ff:ff:ff:ff:ff
5: usb0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 02:22:78:0d:f6:df brd ff:ff:ff:ff:ff:ff
    inet 192.168.103.100/24 brd 192.168.103.255 scope global noprefixroute usb0
       valid_lft forever preferred_lft forever
    inet6 fe80::2d12:dfdd:6c07:530a/64 scope link tentative noprefixroute 
       valid_lft forever preferred_lft forever
6: usb1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether 02:22:78:0d:f6:de brd ff:ff:ff:ff:ff:ff
    inet 192.168.102.2/24 brd 192.168.102.255 scope global noprefixroute usb1
       valid_lft forever preferred_lft forever
    inet6 fe80::549d:3a3d:cd32:1636/64 scope link tentative noprefixroute 
       valid_lft forever preferred_lft forever
mendel@zippy-tang:~$ ping 192.168.103.10
PING 192.168.103.10 (192.168.103.10) 56(84) bytes of data.
From 192.168.103.100 icmp_seq=1 Destination Host Unreachable
From 192.168.103.100 icmp_seq=2 Destination Host Unreachable
From 192.168.103.100 icmp_seq=3 Destination Host Unreachable
^C
--- 192.168.103.10 ping statistics ---
4 packets transmitted, 0 received, +3 errors, 100% packet loss, time 83ms
```

Here is the output of netstat as well
```
mendel@zippy-tang:~$ sudo netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.178.1   0.0.0.0         UG        0 0          0 eth0
0.0.0.0         192.168.178.1   0.0.0.0         UG        0 0          0 wlan0
192.168.102.0   0.0.0.0         255.255.255.0   U         0 0          0 usb1
192.168.103.0   0.0.0.0         255.255.255.0   U         0 0          0 usb0
192.168.178.0   0.0.0.0         255.255.255.0   U         0 0          0 eth0
192.168.178.0   0.0.0.0         255.255.255.0   U         0 0          0 wlan0
```

Not sure if what I'm doing is completely wrong so I figured it would be best to ask for any help.

### Issue Type

Support

### Operating System

Mendel Linux

### Coral Device

Dev Board, Dev Board Micro

hjonnala, 2023-05-02:
On dev board please try the below steps to connect to [dev board micro](https://coral.ai/docs/dev-board-micro/serial-console/#connect-over-usb-otg). Thanks!!

```
sudo apt-get install screen
sudo usermod -aG plugdev,dialout $USER
sudo reboot now
screen /dev/ttyACM0 115200
```

dylan-leslie, 2023-05-03:
Hi @hjonnala, thanks for your response. Apologies maybe I should've been more specific, but I would specifically like to establish a connection with the microcontroller using the USB network interfaces on the coral dev board. Such as, `usb0 usb1 usb2` etc.

`screen` will help me connect to the board and monitor the serial output but doesn't help me for my problem.

I noticed yesterday some differences between my laptop (running ubuntu 20) and the coral dev board for the USB ethernet interfaces. The main difference being the output of `cat /sys/kernel/debug/usb/devices` for both devices like follows:
laptop:
`I:* If#= 2 Alt= 0 #EPs= 2 Cls=02(comm.) Sub=0c Prot=07 Driver=cdc_eem`
coral dev board:
`I:* If#= 2 Alt= 0 #EPs= 2 Cls=02(comm.) Sub=0c Prot=07 Driver=(none)`

This led me to find this on the coral dev board after running `zcat /proc/config.gz | grep -i 'cdc_eem'`
```
# CONFIG_USB_NET_CDC_EEM is not set
```
Is this what I would need to set to allow the coral dev board to communicate with the coral micro over the usb ethernet interface?

Thanks again for your help.