google-coral/coralmicro #39: We need more kindness about examples

cugfk, 2023-05-18:
### Description

Thank you for making good examples but we need more kind explanations.
https://github.com/google-coral/coralmicro/tree/main/examples
That's not easy for beginner how to use and install.

hjonnala, 2023-05-18:
Hi @cugfk could you please follow the documentation on below links, and let me know if you have any issue in trying out any particular example. Thanks..

- https://coral.ai/docs/dev-board-micro/get-started/
- https://github.com/google-coral/coralmicro#get-the-code

cugfk, 2023-05-23:
I need how to start and use more kind texts & pictures for  https://github.com/google-coral/coralmicro/tree/main/examples .

hjonnala, 2023-05-24:
Unforutanely, its not possible due to the other proriorites. Any contributions from communtiy are welcome and greatly appreciated.. Thanks!!

closed 2023-05-24 by hjonnala