google-coral/coralmicro #40: Coral Micro Inference Times Significantly Longer Than USB Stick

drummerhul, 2023-05-19:
### Description

We've been doing some quick and dirty profiling of the stock object detection models on the coral.ai [website](https://coral.ai/models/object-detection/) using `TimerMillis()`. In particular, we've noticed the [SSD MobileNet V2](https://raw.githubusercontent.com/google-coral/test_data/master/tf2_ssd_mobilenet_v2_coco17_ptq_edgetpu.tflite) runs around 83 ms. In previous testing, [Mobile Net V2](
https://github.com/google-coral/edgetpu/raw/master/test_data/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite) running on the USB Accelerator connected to a RPI4b through its USB3 running Python the inference times in the 23 ms range. 

The benchmark on the model page shows 7.6 ms. 

Our expectation was to get close to the same performance (or slightly degraded) as on other host platforms. Though not a 4x difference. 

Is this level of performance degradation on the Coral Micro expected? If so, why?

### Issue Type

Performance

### Operating System

Ubuntu

### Coral Device

USB Accelerator, Dev Board Micro

### Other Devices

Rapsberry Pi 4

### Programming Language

C++, Python 3.9

### Relevant Log Output

```shell
Here's some sample `printf` output on the Coral Micro:

Inference time : 324 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 84 ms 
Inference time : 83 ms 
Inference time : 83 ms 
Inference time : 84 ms 
Inference time : 83 ms
```

hjonnala, 2023-05-20:
> Is this level of performance degradation on the Coral Micro expected? If so, why?

yes, it is expected.

[image]

[image]

drummerhul, 2023-05-22:
To clarify, both performance measurements were running on the Edge TPU, not the MCU or CPU. The above discussion is a generic explanation of why models run slower on MCUs vs CPUs. But in our case, both models are accelerated by the Edge TPU. 

On the Coral Micro here is the code used to setup the Edge TPU Context: 
``` c++
 auto tpu_context = EdgeTpuManager::GetSingleton()->OpenDevice();
  if (!tpu_context) 
  {
    printf("ERROR: Failed to get EdgeTpu context\r\n");
    vTaskSuspend(nullptr);
  }
```

For reference, I did test the [CPU equivalent](https://raw.githubusercontent.com/google-coral/test_data/master/tf2_ssd_mobilenet_v2_coco17_ptq.tflite) on the RT 1176 (M7), Here is a sampling of the results:
``` 
Inference time : 11460 ms 
Inference time : 11458 ms 
Inference time : 11458 ms 
Inference time : 11459 ms 
Inference time : 11458 ms 
Inference time : 11457 ms 
Inference time : 11461 ms 
```
I would appreciate some help understanding why the accelerator performance is so much worse on the Coral Micro vs an RPI4 with a USB Accelerator Stick. Also, why does the benchmark for this model say `7.6ms` when we get `83ms` for the same model?
[image]

Thanks for your help.

hjonnala, 2023-05-22:
[image]

In the above model only edgetpu-custom-op would run on TPU. Dequantize and Tflite_Detection_post_process ops would on CPU. 

> why does the benchmark for this model say `7.6ms` when we get `83ms` for the same model?

The differnce is due to change in host machines( Desktop CPU vs RPI4 CPU). Hope this helps..Thanks!!

hjonnala, 2023-05-24:
> Also, how do we profile the layers embedded within `edgetpu-custom-op` while executing on the Edge TPU?

Its not possible to profile the layers embedded with edgetpu-cusstom-op.

> How can we get `tflite::MicroProfiler` class to correctly output execution times?

Please check with [tflite-micro](https://github.com/tensorflow/tflite-micro) team on how to profile a tflite model. Thanks!!

closed 2023-06-05 by hjonnala