google-coral/coralmicro #41: Coral Dev Board Micro network via USB not work

Xusniyor, 2023-05-28:
### Description

Hello everyone, i installed a virtual machine on windows computer to compile coralmicro sdk and i installed ubuntu 22.04.2 LTS operating system on virtual machine, coralmicro sdk is compilable and the software is loaded on Coral Dev Board Micro only there is one problem Coral Dev Board Micro internet via USB not working.

[image]

[image]

### Issue Type

Feature Request

### Operating System

Windows 10, Linux, Ubuntu

### Coral Device

Dev Board Micro

### Programming Language

C++

### Relevant Log Output

```shell
Traceback (most recent call last):
  File "/home/xusi/CoralDevBoardMicro/coralmicro/examples/audio_streaming/audio_client.py", line 322, in main
    sock.connect((args.host, args.port))
TimeoutError: timed out

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/xusi/CoralDevBoardMicro/coralmicro/examples/audio_streaming/audio_client.py", line 356, in <module>
    main()
  File "/home/xusi/CoralDevBoardMicro/coralmicro/examples/audio_streaming/audio_client.py", line 328, in main
    raise RuntimeError(msg) from e
RuntimeError: Cannot connect to Coral Dev Board Micro, make sure you specify the correct IP address with --host.
```

hjonnala, 2023-05-30:
@Xusniyor we don't have support for virtual machines. On windows, please try the [Arduino IDE 2.0.4](https://github.com/arduino/arduino-ide/releases/tag/2.0.4) to flash the coral micro examples. Thanks!!

dadanugm, 2023-09-29:
Hi @hjonnala . Got the same issue here
[image]

flashing done through COM13 on Arduino IDE, but what is that Coral Dev Board Micro (as not recognized device)?
Do we have driver for that on Windows?
I believe that for accessing the flash content. I tried camera example code, but how to view it? the images stored in Coral Micro Flash memory.

rslavin, 2023-10-26:
I am having the same issue. I can't get any of the camera code to work over USB. Running `camera_streaming_app.py` displays Chrome but I eventually get a ConnectTimeout exception for host='10.10.10.1'. Watching the serial monitor over UART, the micro prints that it is starting the PRC server on that IP address, but I don't understand how to connect to the address over USB.

Pratyush-Mallick, 2023-11-08:
I'm trying to connect the coral board to the raspberry pi 4, but the hard coded IP 10.10.10.1 doesn't get detected on the USB.