google-coral/coralmicro #44: Get Started for Mac M2

ramkumarkoppu, 2023-06-13:
### Description

By following instruction in https://coral.ai/docs/dev-board-micro/get-started/#5-try-the-face-detection-example on Apple Silicon Mac, I get following error:
bash-3.2$ lsusb
Bus 000 Device 002: ID 18d1:9308 Google Inc. Coral Dev Board Micro  Serial: 1b31080e829089dd
Bus 000 Device 000: ID 18d1:9308 Google Inc. USB 3.1 Bus 
bash-3.2$ lsusb
Bus 000 Device 002: ID 18d1:9308 Google Inc. Coral Dev Board Micro  Serial: 1b31080e829089dd
Bus 000 Device 000: ID 18d1:9308 Google Inc. USB 3.1 Bus 
bash-3.2$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
	Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x0001780c | crc=0x852c658e | flg=0x0110
  LOAD | adr=0x00000008 | len=0x0001780c | crc=0xcbf635c8 | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
**Python interpreter cannot find libusb, please install it.**

### Issue Type

Build/Install

### Operating System

Mac OS

### Coral Device

Dev Board Micro

### Programming Language

Python 3.9

### Relevant Log Output

```shell
bash-3.2$ lsusb
Bus 000 Device 002: ID 18d1:9308 Google Inc. Coral Dev Board Micro Serial: 1b31080e829089dd
Bus 000 Device 000: ID 18d1:9308 Google Inc. USB 3.1 Bus
bash-3.2$ lsusb
Bus 000 Device 002: ID 18d1:9308 Google Inc. Coral Dev Board Micro Serial: 1b31080e829089dd
Bus 000 Device 000: ID 18d1:9308 Google Inc. USB 3.1 Bus
bash-3.2$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
ENA | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
LOAD | adr=0x00000004 | len=0x0001780c | crc=0x852c658e | flg=0x0110
LOAD | adr=0x00000008 | len=0x0001780c | crc=0xcbf635c8 | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
Python interpreter cannot find libusb, please install it.
```

hjonnala, 2023-06-13:
please try the steps mentione [here](https://github.com/google-coral/coralmicro/blob/main/scripts/flashtool.py#L1197) to install libusb. Thanks!!

ramkumarkoppu, 2023-06-13:
Thanks tried it, I get,

bash-3.2$ brew install libusb
==> Downloading https://formulae.brew.sh/api/formula.jws.json

==> Downloading https://formulae.brew.sh/api/cask.jws.json

Warning: libusb 1.0.26 is already installed and up-to-date.
To reinstall 1.0.26, run:
  brew reinstall libusb
bash-3.2$ sudo ln -s $(brew --prefix)/lib/libusb-1.0.0.dylib /usr/local/lib/libusb.dylib
ln: /usr/local/lib/libusb.dylib: No such file or directory
bash-3.2$

hjonnala, 2023-06-13:
Please share the output of below command:

brew info libusb

ramkumarkoppu, 2023-06-13:
(tensorflow-macos) bash-3.2$ brew info libusb
==> libusb: stable 1.0.26 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/1.0.26 (22 files, 595.0KB) *
  Poured from bottle using the formulae.brew.sh API on 2023-06-13 at 21:10:42
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 31,865 (30 days), 6,365 (90 days), 838,148 (365 days)
install-on-request: 6,083 (30 days), 1,620 (90 days), 84,598 (365 days)
build-error: 0 (30 days)
(tensorflow-macos) bash-3.2$

hjonnala, 2023-06-14:
Looks like the system python interpreter cannot find the libusb installed by brew. Can you try  to install python via brew.

ramkumarkoppu, 2023-06-14:
Hi Hemanth (@hjonnala) you are a star contributor for this community. Thank you for your help.
As per your recommendation, I followed below steps and did work on my MacBookPro M2 Max:

bash-3.2$ `brew install python`
==> Fetching python@3.11
Python has been installed as
  /opt/homebrew/bin/python3

bash-3.2$ `which python3`
/opt/homebrew/bin/python3

bash-3.2$ `python3 --version`
Python 3.11.4

bash-3.2$ `brew uninstall libusb`
Uninstalling /opt/homebrew/Cellar/libusb/1.0.26... (22 files, 595.0KB)

bash-3.2$ `brew install libusb`
bash-3.2$ brew info libusb
==> libusb: stable 1.0.26 (bottled), HEAD
Library for USB device access
https://libusb.info/
/opt/homebrew/Cellar/libusb/1.0.26 (22 files, 595.0KB) *
  Poured from bottle using the formulae.brew.sh API on 2023-06-14 at 21:11:33
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/libusb.rb
License: LGPL-2.1-or-later
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 30,951 (30 days), 6,321 (90 days), 833,064 (365 days)
install-on-request: 6,083 (30 days), 1,614 (90 days), 84,159 (365 days)
build-error: 0 (30 days)

bash-3.2$ `lsusb`
Bus 000 Device 002: ID 18d1:9308 Google Inc. Coral Dev Board Micro  Serial: 1b31080e829089dd
Bus 000 Device 000: ID 18d1:9308 Google Inc. USB 3.1 Bus 

bash-3.2$ `bash setup.sh`

bash-3.2$ `bash build.sh`

bash-3.2$ `python3 scripts/flashtool.py -e detect_faces`

Hopefully this will help someone like me who using Apple Silicon Mac. Cheers!

closed 2023-06-14 by ramkumarkoppu

cyborgdennett, 2023-09-14:
Hi, I have the same issue on WSL2(ubuntu 22)

coralmicro$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
        Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x00017854 | crc=0xa4589820 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x00017854 | crc=0xc57210fa | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
Encountered an error during flashing: Unable to find device in SDP mode.

cyborgdennett, 2023-09-14:
Hi, I have the same issue on Windows (WSL) ubuntu22 

```bash
coralmicro$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 18d1:9308 Google Inc. Coral Dev Board Micro
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

coralmicro$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
        Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x00017854 | crc=0xdd594538 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x00017854 | crc=0x9b88b39a | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
Encountered an error during flashing: Unable to find device in SDP mode.
```

cyborgdennett, 2023-09-14:
So one solution I found is to spam in powershell:
```powershell
usbipd wsl attach --busid 1-2
```
Then it will go through this step.

next error is as follow:
```bash
coralmicro$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
        Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x00017854 | crc=0x93bf7879 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x00017854 | crc=0xf5171da3 | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
STATE_LOAD_FLASHLOADER
Traceback (most recent call last):
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 1193, in <module>
    main()
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 1182, in main
    RunFlashtool(StateCheckForAny, sbfile_path=sbfile_path,
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 821, in RunFlashtool
    (prev_state, state) = (state, state(**state_kwargs))
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 543, in StateLoadFlashloader
    subprocess.check_call([blhost_path, '-u', sdp_vidpid(), '--', 'load-image',
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/casper/ml/coralmicro/third_party/nxp/blhost/bin/linux/amd64/blhost', '-u', '0x1fc9,0x13d', '--', 'load-image', '/tmp/tmpudxc1pku/ivt_flashloader.bin']' returned non-zero exit status 1.
```

cyborgdennett, 2023-09-14:
You can use 
```powershell
usbipd wsl attach --busid 1-2 -a
```
The -a will make the usbipd service to keep looking for the busid even after it is disonnected for a short while.

cyborgdennett, 2023-09-14:
I tried updating the blhost binary, but same error