google-coral/coralmicro #45: No serial device detected after changing example using Arduino IDE

arthurkafer, 2023-06-20:
### Description

I'm using Arduino IDE 2.0.4 to flash a custom code based on the DetectObjectsFile example. I bought 10 CoralMicro boards to test them but sometimes flashing the code just makes them unacessible. 

No serial port is opened when I connect the board and even clicking the User Button to enter SDP mode does not work, it doesnt recognize anything.

Important to mention that I can work with the board for a considerable amount of time, but in some code flashings the board simply doesn't respond or even open serial connections.

Is there a way to reflash something by serial or anything to make it respond again?

I already have 3 of those 10 boards in this state, and flashing the same exact firmware in another one works fine, but I don't want to lose them all trying this method.

### Issue Type

Bug, Build/Install

### Operating System

Windows 10

### Coral Device

Dev Board Micro

### Programming Language

Python 3.9

arthurkafer, 2023-06-20:
After the bug occurs
[image]

Before the bug occurs
[image]

arthurkafer, 2023-06-21:
Using windows 10 there's no board detection and no serial port opened.

Right now I can only test the boards in windows machines, but putting it into SDP mode (like mentioned at that comment) does nothing.
It does not connect to any serial port and the board is not detected using USB listing commands.

[image]

I'll continue testing and developing the code and when the error occurs again I'll share the code and flash logs. If I remember correctly, flash logs are fine and without any errors, but the board does not restart with the new code (actually it doesn't start at all)

hjonnala, 2023-06-21:
With the model you provided I am able to falsh and watch the serial console logs. I don't see model loading section in your flash logs. please check the sketech folder and place model inside data folder.  Thanks!!!

Look like the model does not work with Dev Board Micro!! There are two subgraphs in the model, can you try with one subgraph?

serial consoel logs:

[image]

flash logs:

```
	Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x000178c4 | crc=0xe89368d6 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x000178c4 | crc=0x849acdbd | flg=0x0110
Inject command 'receive-sb-file'
Preparing to send 193264 (0x2f2f0) bytes to the target.
Successful generic response to command 'receive-sb-file'

(1/1) 0%
(1/1) 1%
(1/1) 1%
(1/1) 2%
(1/1) 2%
(1/1) 3%
(1/1) 3%
(1/1) 4%
(1/1) 4%
(1/1) 5%
(1/1) 5%
(1/1) 6%
(1/1) 6%
(1/1) 7%
(1/1) 7%
(1/1) 8%
(1/1) 8%
(1/1) 9%
(1/1) 9%
(1/1)10%
(1/1)10%
(1/1)11%
(1/1)12%
(1/1)12%
(1/1)13%
(1/1)13%
(1/1)14%
(1/1)14%
(1/1)15%
(1/1)15%
(1/1)16%
(1/1)16%
(1/1)17%
(1/1)17%
(1/1)18%
(1/1)18%
(1/1)19%
(1/1)19%
(1/1)20%
(1/1)20%
(1/1)21%
(1/1)21%
(1/1)22%
(1/1)23%
(1/1)23%
(1/1)24%
(1/1)24%
(1/1)25%
(1/1)25%
(1/1)26%
(1/1)26%
(1/1)27%
(1/1)27%
(1/1)28%
(1/1)28%
(1/1)29%
(1/1)29%
(1/1)30%
(1/1)30%
(1/1)31%
(1/1)31%
(1/1)32%
(1/1)32%
(1/1)33%
(1/1)34%
(1/1)34%
(1/1)35%
(1/1)35%
(1/1)36%
(1/1)36%
(1/1)37%
(1/1)37%
(1/1)38%
(1/1)38%
(1/1)39%
(1/1)39%
(1/1)40%
(1/1)40%
(1/1)41%
(1/1)41%
(1/1)42%
(1/1)42%
(1/1)43%
(1/1)43%
(1/1)44%
(1/1)45%
(1/1)45%
(1/1)46%
(1/1)46%
(1/1)47%
(1/1)47%
(1/1)48%
(1/1)48%
(1/1)49%
(1/1)49%
(1/1)50%
(1/1)50%
(1/1)51%
(1/1)51%
(1/1)52%
(1/1)52%
(1/1)53%
(1/1)53%
(1/1)54%
(1/1)54%
(1/1)55%
(1/1)56%
(1/1)56%
(1/1)57%
(1/1)57%
(1/1)58%
(1/1)58%
(1/1)59%
(1/1)59%
(1/1)60%
(1/1)60%
(1/1)61%
(1/1)61%
(1/1)62%
(1/1)62%
(1/1)63%
(1/1)63%
(1/1)64%
(1/1)64%
(1/1)65%
(1/1)65%
(1/1)66%
(1/1)67%
(1/1)67%
(1/1)68%
(1/1)68%
(1/1)69%
(1/1)69%
(1/1)70%
(1/1)70%
(1/1)71%
(1/1)71%
(1/1)72%
(1/1)72%
(1/1)73%
(1/1)73%
(1/1)74%
(1/1)74%
(1/1)75%
(1/1)75%
(1/1)76%
(1/1)76%
(1/1)77%
(1/1)78%
(1/1)78%
(1/1)79%
(1/1)79%
(1/1)80%
(1/1)80%
(1/1)81%
(1/1)81%
(1/1)82%
(1/1)82%
(1/1)83%
(1/1)83%
(1/1)84%
(1/1)84%
(1/1)85%
(1/1)85%
(1/1)86%
(1/1)86%
(1/1)87%
(1/1)87%
(1/1)88%
(1/1)89%
(1/1)89%
(1/1)90%
(1/1)90%
(1/1)91%
(1/1)91%
(1/1)92%
(1/1)92%
(1/1)93%
(1/1)93%
(1/1)94%
(1/1)94%
(1/1)95%
(1/1)95%
(1/1)96%
(1/1)96%
(1/1)97%
(1/1)97%
(1/1)98%
(1/1)98%
(1/1)99%
(1/1)100% Completed!
Successful generic response to command 'receive-sb-file'
Response status = 0 (0x0) Success.
Wrote 193264 of 193264 bytes.
Finding all necessary files
Creating Filesystem
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
STATE_LOAD_FLASHLOADER
STATE_CHECK_FOR_FLASHLOADER
STATE_PROGRAM
STATE_LOAD_ELFLOADER
STATE_PROGRAM_DATA_FILES
OpenHidDevice vid=18d1 pid=9307 ...
OpenHidDevice vid=18d1 pid=9307 opened
Transferring /highwayframe.rgb
/highwayframe.rgb 10%
/highwayframe.rgb 20%
/highwayframe.rgb 30%
/highwayframe.rgb 40%
/highwayframe.rgb 50%
/highwayframe.rgb 60%
/highwayframe.rgb 70%
/highwayframe.rgb 80%
/highwayframe.rgb 90%
/highwayframe.rgb done
Transferring /models/yolov5s-int8-224_edgetpu.tflite
/models/yolov5s-int8-224_edgetpu.tflite 10%
/models/yolov5s-int8-224_edgetpu.tflite 20%
/models/yolov5s-int8-224_edgetpu.tflite 30%
/models/yolov5s-int8-224_edgetpu.tflite 40%
/models/yolov5s-int8-224_edgetpu.tflite 50%
/models/yolov5s-int8-224_edgetpu.tflite 60%
/models/yolov5s-int8-224_edgetpu.tflite 70%
/models/yolov5s-int8-224_edgetpu.tflite 80%
/models/yolov5s-int8-224_edgetpu.tflite 90%
/models/yolov5s-int8-224_edgetpu.tflite done
Transferring /default.elf
/default.elf 10%
/default.elf 20%
/default.elf 30%
/default.elf 40%
/default.elf 50%
/default.elf 60%
/default.elf 70%
/default.elf 80%
/default.elf 90%
/default.elf done
Transferring /usb_ip_address
/usb_ip_address done
STATE_RESET_TO_FLASH
OpenHidDevice vid=18d1 pid=9307 ...
OpenHidDevice vid=18d1 pid=9307 opened
Flashing to flash storage complete, the device is restarting to execute your application.

```

arthurkafer, 2023-06-21:
Using static images (files) the model works fine, try this example out:

[example.zip](https://github.com/google-coral/coralmicro/files/11820193/example.zip)

The strange thing is that you flashed the example fine and most of the time I can do the same thing, but sometimes these random board crashes happen

hjonnala, 2023-06-21:
It is also working fine. Try resetting the device by plugging usb into Dev board Micro instead host machine whie pressing the user button.

[image]

arthurkafer, 2023-06-21:
Yes, I know it works fine, that's why I asked about the code failing to run when I use the camera.

As I said before, the Dev board micro does not reset when I plug it pressing the user button. Is there another way to reset it or to force a serial connection to flash code again?

hjonnala, 2023-06-21:
> As I said before, the Dev board micro does not reset when I plug it pressing the user button. Is there another way to reset it or to force a serial connection to flash code again?

https://coral.ai/docs/dev-board-micro/arduino/#troubleshooting
[image]

hjonnala, 2023-06-21:
> Testing using SDP mode (user button pressed when connecting USB cable, and releasing after connection):
> 
>  VID_20230621_100007.1.mp4

Please try resetting the device by plugging usb into Dev board Micro instead host machine whie pressing the user button.

arthurkafer, 2023-06-21:
> > As I said before, the Dev board micro does not reset when I plug it pressing the user button. Is there another way to reset it or to force a serial connection to flash code again?
> 
> https://coral.ai/docs/dev-board-micro/arduino/#troubleshooting [image]

I'll provide videos doing that, the result is the same.

Pluggin USB into Dev board insted of the host machine while pressing the user button:
https://github.com/google-coral/coralmicro/assets/49439463/d257c188-88fe-40a6-829b-3f747b23cf9f

Holding user button while I press the reset button:
https://github.com/google-coral/coralmicro/assets/49439463/4ec268a4-a827-428d-820a-3e31be2b7c7f

hjonnala, 2023-06-21:
ok, please try with any Linux machine if avialable. Thanks!!

arthurkafer, 2023-06-21:
Ok, I installed dual boot here and now I can try to run it in a Ubuntu 22.

The serial port still doesn't show up, but at least I can see the board in lsusb.

Video of pressing user button in linux, recongnized in lsusb, serial port not found:
https://github.com/google-coral/coralmicro/assets/49439463/cbde051a-1ecd-4ab6-85e6-3466ddf11c67

Flash logs trying to flash code right after the video (using SDP Mode, user button pressed when connected)
```
Traceback (most recent call last):
  File "flashtool.py", line 1193, in <module>
  File "flashtool.py", line 1043, in main
  File "flashtool.py", line 821, in RunFlashtool
  File "flashtool.py", line 543, in StateLoadFlashloader
  File "subprocess.py", line 364, in check_call
subprocess.CalledProcessError: Command '['/tmp/_MEIIEzABq/third_party/nxp/blhost/bin/linux/amd64/blhost', '-u', '0x1fc9,0x13d', '--', 'load-image', '/tmp/tmp_kv0gwkx/ivt_flashloader.bin']' returned non-zero exit status 1.
[7253] Failed to execute script 'flashtool' due to unhandled exception!
Failed uploading: uploading error: exit status 1

```

Video trying a normal flash, but the error is board not found (flash log in the video):

https://github.com/google-coral/coralmicro/assets/49439463/7b7db188-aee7-4b7c-8d7d-74a996839764

hjonnala, 2023-06-21:
flahstool script looks for this pattern: https://github.com/google-coral/coralmicro/blob/main/scripts/flashtool.py#L209. But in your video it i showing 18D1:9307..

Can you please try the blink led example with RTOS on your ubuntu machine: https://github.com/google-coral/coralmicro/tree/main

arthurkafer, 2023-06-22:
I've tried to flash the blink led example in the board and it is alive back again. Thanks for your help.

Is there a way to make the board work again using only Windows OS? Because I cant reflash everytime those boards using Linux when they crash.

If there isn't a way already developed, can you guide me a way to use windows and reflash it?

closed 2023-06-22 by arthurkafer

bogdannedelcu, 2023-09-30:
I have some similar issue here, if you could provide any clues... thank you

https://github.com/google-coral/coralmicro/issues/74

rslavin, 2023-10-12:
I have the same issue, only lsusb doesn't show anything. I've held all combination of buttons while plugging it in and all I get is the amber light and no new device detected.