google-coral/coralmicro #46: Understanding the error about the model

arthurkafer, 2023-06-22:
With the model you provided I am able to falsh and watch the serial console logs. I don't see model loading section in your flash logs. please check the sketech folder and place model inside data folder.  Thanks!!!

Look like the model does not work with Dev Board Micro!! There are two subgraphs in the model, can you try with one subgraph?

serial consoel logs:

[image]

flash logs:

```
	Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x000178c4 | crc=0xe89368d6 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x000178c4 | crc=0x849acdbd | flg=0x0110
Inject command 'receive-sb-file'
Preparing to send 193264 (0x2f2f0) bytes to the target.
Successful generic response to command 'receive-sb-file'

(1/1) 0%
(1/1) 1%
(1/1) 1%
(1/1) 2%
(1/1) 2%
(1/1) 3%
(1/1) 3%
(1/1) 4%
(1/1) 4%
(1/1) 5%
(1/1) 5%
(1/1) 6%
(1/1) 6%
(1/1) 7%
(1/1) 7%
(1/1) 8%
(1/1) 8%
(1/1) 9%
(1/1) 9%
(1/1)10%
(1/1)10%
(1/1)11%
(1/1)12%
(1/1)12%
(1/1)13%
(1/1)13%
(1/1)14%
(1/1)14%
(1/1)15%
(1/1)15%
(1/1)16%
(1/1)16%
(1/1)17%
(1/1)17%
(1/1)18%
(1/1)18%
(1/1)19%
(1/1)19%
(1/1)20%
(1/1)20%
(1/1)21%
(1/1)21%
(1/1)22%
(1/1)23%
(1/1)23%
(1/1)24%
(1/1)24%
(1/1)25%
(1/1)25%
(1/1)26%
(1/1)26%
(1/1)27%
(1/1)27%
(1/1)28%
(1/1)28%
(1/1)29%
(1/1)29%
(1/1)30%
(1/1)30%
(1/1)31%
(1/1)31%
(1/1)32%
(1/1)32%
(1/1)33%
(1/1)34%
(1/1)34%
(1/1)35%
(1/1)35%
(1/1)36%
(1/1)36%
(1/1)37%
(1/1)37%
(1/1)38%
(1/1)38%
(1/1)39%
(1/1)39%
(1/1)40%
(1/1)40%
(1/1)41%
(1/1)41%
(1/1)42%
(1/1)42%
(1/1)43%
(1/1)43%
(1/1)44%
(1/1)45%
(1/1)45%
(1/1)46%
(1/1)46%
(1/1)47%
(1/1)47%
(1/1)48%
(1/1)48%
(1/1)49%
(1/1)49%
(1/1)50%
(1/1)50%
(1/1)51%
(1/1)51%
(1/1)52%
(1/1)52%
(1/1)53%
(1/1)53%
(1/1)54%
(1/1)54%
(1/1)55%
(1/1)56%
(1/1)56%
(1/1)57%
(1/1)57%
(1/1)58%
(1/1)58%
(1/1)59%
(1/1)59%
(1/1)60%
(1/1)60%
(1/1)61%
(1/1)61%
(1/1)62%
(1/1)62%
(1/1)63%
(1/1)63%
(1/1)64%
(1/1)64%
(1/1)65%
(1/1)65%
(1/1)66%
(1/1)67%
(1/1)67%
(1/1)68%
(1/1)68%
(1/1)69%
(1/1)69%
(1/1)70%
(1/1)70%
(1/1)71%
(1/1)71%
(1/1)72%
(1/1)72%
(1/1)73%
(1/1)73%
(1/1)74%
(1/1)74%
(1/1)75%
(1/1)75%
(1/1)76%
(1/1)76%
(1/1)77%
(1/1)78%
(1/1)78%
(1/1)79%
(1/1)79%
(1/1)80%
(1/1)80%
(1/1)81%
(1/1)81%
(1/1)82%
(1/1)82%
(1/1)83%
(1/1)83%
(1/1)84%
(1/1)84%
(1/1)85%
(1/1)85%
(1/1)86%
(1/1)86%
(1/1)87%
(1/1)87%
(1/1)88%
(1/1)89%
(1/1)89%
(1/1)90%
(1/1)90%
(1/1)91%
(1/1)91%
(1/1)92%
(1/1)92%
(1/1)93%
(1/1)93%
(1/1)94%
(1/1)94%
(1/1)95%
(1/1)95%
(1/1)96%
(1/1)96%
(1/1)97%
(1/1)97%
(1/1)98%
(1/1)98%
(1/1)99%
(1/1)100% Completed!
Successful generic response to command 'receive-sb-file'
Response status = 0 (0x0) Success.
Wrote 193264 of 193264 bytes.
Finding all necessary files
Creating Filesystem
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
STATE_LOAD_FLASHLOADER
STATE_CHECK_FOR_FLASHLOADER
STATE_PROGRAM
STATE_LOAD_ELFLOADER
STATE_PROGRAM_DATA_FILES
OpenHidDevice vid=18d1 pid=9307 ...
OpenHidDevice vid=18d1 pid=9307 opened
Transferring /highwayframe.rgb
/highwayframe.rgb 10%
/highwayframe.rgb 20%
/highwayframe.rgb 30%
/highwayframe.rgb 40%
/highwayframe.rgb 50%
/highwayframe.rgb 60%
/highwayframe.rgb 70%
/highwayframe.rgb 80%
/highwayframe.rgb 90%
/highwayframe.rgb done
Transferring /models/yolov5s-int8-224_edgetpu.tflite
/models/yolov5s-int8-224_edgetpu.tflite 10%
/models/yolov5s-int8-224_edgetpu.tflite 20%
/models/yolov5s-int8-224_edgetpu.tflite 30%
/models/yolov5s-int8-224_edgetpu.tflite 40%
/models/yolov5s-int8-224_edgetpu.tflite 50%
/models/yolov5s-int8-224_edgetpu.tflite 60%
/models/yolov5s-int8-224_edgetpu.tflite 70%
/models/yolov5s-int8-224_edgetpu.tflite 80%
/models/yolov5s-int8-224_edgetpu.tflite 90%
/models/yolov5s-int8-224_edgetpu.tflite done
Transferring /default.elf
/default.elf 10%
/default.elf 20%
/default.elf 30%
/default.elf 40%
/default.elf 50%
/default.elf 60%
/default.elf 70%
/default.elf 80%
/default.elf 90%
/default.elf done
Transferring /usb_ip_address
/usb_ip_address done
STATE_RESET_TO_FLASH
OpenHidDevice vid=18d1 pid=9307 ...
OpenHidDevice vid=18d1 pid=9307 opened
Flashing to flash storage complete, the device is restarting to execute your application.

```

_Originally posted by @hjonnala in https://github.com/google-coral/coralmicro/issues/45#issuecomment-1601108395_

hjonnala, 2023-06-23:
I haven't seen that error before. Can you try the same script with the different models..

- Different edgetpu model from coral example.
- uncompiled yolo tflite model
- edgetpu compiled model with only one subgraph (i.e compile the model without -a flag)

arthurkafer, 2023-06-23:
Hello, I solved it using an interesting method, but I don't think it is running the model correctly (at least it runs now).

The model loading error happens when I pass the grabbed image directly to the input_tensor:
```
    if (Camera.grab(tflite::GetTensorData<uint8_t>(input_tensor)) != CameraStatus::SUCCESS) {
      Serial.println("cannot invoke because camera failed to grab frame");
      return;
    }
```

Doing like this, it actually runs the model, I just need to check if the image is being inferenced the right way:
```
// creating frame_buffer inside the namespace
FrameBuffer frame_buffer;

// in while loop
    if (Camera.grab(frame_buffer) != CameraStatus::SUCCESS) {
      Serial.println("cannot invoke because camera failed to grab frame");
      return;
    }
    std::memcpy(tflite::GetTensorData<uint8_t>(input_tensor), frame_buffer.getBuffer(), frame_buffer.getBufferSize());
```

One thing I noted is that I was using raw rgb images when testing the model inference, but the _I think_ the camera is grabbing a BGR formatted image (even if I created the Camera instance using `coralmicro::CameraFormat::kRgb`).

I'll try it out too.

arthurkafer, 2023-06-27:
I can't be 100% sure about the problem that happens using the camera directly, but I suspect the first frames are not valid and then the neural network buffer does not match with the image dimensions.

The image format using kRgb works fine, I was using it the wrong way.

Thanks for the responses, I'll keep this thread updated if I find something out.

closed 2023-06-27 by arthurkafer