google-coral/coralmicro #47: Data folder is not flashed using Coral Dev Micro + Wireless Add-on board

arthurkafer, 2023-06-29:
### Description

Good afternoon,

I'm trying to create a simple HTTP server using Wi-Fi (and the add-on board). Code flashing is fine, but I suspect the /data files in the project are not being flashed to the board when the Dev Board Micro + Wireless Add-on is set on Arduino IDE.

Using the normal Dev Board Micro flashing the server works as intended, because the /data folder is flashed correctly.

Everytime I'm flashing using Sketch+Data option:
[image]

I'll put the code I'm using and the flash logs for both boards below.

Dev Board Micro + Wireless Add-on -> [CameraStreamingHttp using WiFi](https://github.com/google-coral/coralmicro/files/11908699/CameraStreamingHttp.zip)

Dev Board Micro -> [CameraStreamingHttp without WiFi.zip](https://github.com/google-coral/coralmicro/files/11908737/CameraStreamingHttp.no.wifi.zip)

Is there something missing in Arduino IDE to flash the /data folder together?

### Issue Type

Bug

### Operating System

Windows 10

### Coral Device

Dev Board Micro

### Programming Language

C++

### Relevant Log Output

```shell
*Selecting "Dev Board Micro + Wireless Add-on" board*

	Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x000178c4 | crc=0xbb66d509 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x000178c4 | crc=0xa1f260bf | flg=0x0110
Inject command 'receive-sb-file'
Preparing to send 193248 (0x2f2e0) bytes to the target.
Successful generic response to command 'receive-sb-file'

(1/1) 0%
(1/1) 1%
(1/1) 1%
(1/1) 2%
(1/1) 2%
(1/1) 3%
(1/1) 3%
(1/1) 4%
(1/1) 4%
(1/1) 5%
(1/1) 5%
(1/1) 6%
(1/1) 6%
[... 468 lines cut ...]
/coral_micro_camera.html 40%
/coral_micro_camera.html 50%
/coral_micro_camera.html 60%
/coral_micro_camera.html 70%
/coral_micro_camera.html 80%
/coral_micro_camera.html 90%
/coral_micro_camera.html done
Transferring /default.elf
/default.elf 10%
/default.elf 20%
/default.elf 30%
/default.elf 40%
/default.elf 50%
/default.elf 60%
/default.elf 70%
/default.elf 80%
/default.elf 90%
/default.elf done
Transferring /usb_ip_address
/usb_ip_address done
STATE_RESET_TO_FLASH
OpenHidDevice vid=18d1 pid=9307 ...
OpenHidDevice vid=18d1 pid=9307 opened
Flashing to flash storage complete, the device is restarting to execute your application.
```

hjonnala, 2023-06-29:
please refer to this comment: https://github.com/google-coral/coralmicro-out-of-tree-sample/issues/6#issuecomment-1515293630

arthurkafer, 2023-06-29:
Yeah, Show Sketch Folder puts me in the correct folder, and the file is there in the /data folder.

[image]
[image]

I found the /data folder is being "replaced"/"overwritten" by the /data folder inside the Arduino core libraries -> C:\Users\arthu\AppData\Local\Arduino15\packages\coral\hardware\coral_micro\1.0.1\variants\coral_micro_wifi\data
[image]

I tried putting the html file inside this folder and it was flashed correctly, but that's not supposed to work that way, I guess

[image]

FStefanni, 2023-07-18:
Hi,

I confirm this bug.
I have done a little analysis, here what I have found:
* In file `boards.txt`, the key `coral_micro_wifi.upload.pattern` passes two times the option `--data_dir`:
    * The first time, as for the `micro` (i.e. without wifi), to pass the tensorflow data model
    * The second time, to pass the `data` dir inside the `variants` folder, I suppose to load some specific firmware to enable the WiFi
    * So, in the end, it seems that passing two times the same option, does not work, since the second occurrence overwrites the first occurrence
* I have run the `flashtool` program, with `--help` to check this fact, but the option `--data_dir` is not listed. i suppose this is a customization of a tihrd-party tool, and who have done the customization has forgotten to describe this option in the help
* To confirm the bug I have done two checks:
    *   If I run the upload as is, then I see in the log that the files inside the `data` dir of `variants` is actually loaded, but not the model inside the project dir
    * If I remove the second occurrence of `--data_dir`, I see the model is loaded, the files inside `data` of `variants` are not loaded, and, in this case, without using the wifi, the tensorflow model works fine

I note the tags of this issue, and I suggest to change them, since it does not work regardless of the platform (I am using MacOS and Linux, the original reported of this issue Windows, and it is normal that this cannot work regardless of the platform). 

Regards