google-coral/coralmicro #51: Cannot Detect Dev Board Micro over USB

Neonel21, 2023-07-30:
### Description

Hello!
I've just recently bought the Dev Board Micro by Google Coral and have little to no experience on it. As I was setting it up per the instructions from the website [Google Coral Setup](https://coral.ai/docs/dev-board-micro/get-started/), I could not find the device listed whilst verification. It does not appear to be verified, and I cannot find it by running lsusb. For context, I am currently running Ubuntu version 22.04.2 on Oracle VM VirtualBox. Prior to this, I have had another Virtual Machine with the same version, in which I could list the dev board; however, with the current machine, I cannot.
I've tried to check whether the issue was on the Ubuntu version, or the VM VirtualBox; however, it does not appear to be so. I've also tried to check the cable and connecting to other device running Windows10 and Windows11, and yet the problem remains.
For further context, whenever I pop in the Dev Board Micro, I initally receive a sound of "USB Connected" notification, but that is immediatelly replaced by another sound of "USB Disconnected" notification. I've also tried checking the USB connection status through Windows Powershell Get-PnpDevice command on both Windows 10 and 11 computers.  In Windows 11, the status is "Error", and in Windows 10, I cannot even view the Dev Board. 
I would like to apologise for taking up your precious time. It would be appreciated If I could get help on this issue. 

### Issue Type

Build/Install, Others

### Operating System

Windows 10, Ubuntu

### Coral Device

Dev Board Micro

### Programming Language

C++, Python 3.8

cyborgdennett, 2023-09-14:
I have the same problem. Please tell me when you found out what to do.

hpssjellis, 2023-09-14:
I have only had my board for 2 days but I have learnt:

1. always push the middle button when connecting the usb cable to program the board, then after programming unplug and plug in the usb cable to run the program. The other methods do not seem to work.

On linux ```lsusb``` shows ```NXP``` when the middle button is pushed and ```Corel``` when the program runs. On Windows when the middle button is pushed the port does not show at all but still uploads code, when no button pushed the port shows up on windows.

I would assume your boards have bad code on them and needs to have good code programmed before anything works except lsusb to show the NXP statement.

cyborgdennett, 2023-09-17:
@hpssjellis how do you flash using windows? I tried with WSL but to no avail

hpssjellis, 2023-09-17:
@cyborgdennett  I havent tried WSL, but generally would not even expect it to work. The Arduino IDE does flash simple programs to the coral-micro. I have not had any success with the ML code yet. Trusting that the arduino IDE can find the port is a bit of a pain. Start off with just a regualr blink or serial monitor code first.

hpssjellis, 2023-09-17:
@cyborgdennett I however cant seem to get serial read working at all, so I may have to swtich to a different board.

cyborgdennett, 2023-09-18:
@hpssjellis in the documentation it says you need an serial to USB port, haven't tried it though.

hpssjellis, 2023-09-18:
> @hpssjellis in the documentation it says you need an serial to USB cable, haven't tried it though.

@cyborgdennett   I remember seeing that and then things kind of worked so I ignored it. Thanks for reminding me. Would that help with WSL?

cyborgdennett, 2023-09-18:
I don't think so.

hpssjellis, 2023-09-18:
@cyborgdennett I will try WSL today. Got a good link for the setup? I haven't used it since it first came out and nothing I needed worked. Also one of my students suggested a virtualbox to setup linux on windows. Is that an option?

cyborgdennett, 2023-09-18:
I also was trying with ubuntu22. The main problem is that wel has trouble with usb. You can download usbipd, as recommended by Microsoft, and then open the port your coral micro is connected to, with
`usbipd WSL attach --busid 1-2 -a` in a administrator Powershell shell. which will connect Linux with the usb device over usb/IP. Then you will still get problems since the vidpod of the coral micro changes on every part of its flashing procedure.

hpssjellis, 2023-09-18:
@cyborgdennett . I tried setting WSL up on a school computer and it is not as easy as it should be. Seems the virtual environment is getting grumpy. Not sure why. If I get it working I will try your steps, but not really expecting it too work. I much prefer an old laptop that is really cheap and installing ubuntu?

hpssjellis, 2023-09-18:
@cyborgdennett THings are getting interesting. WSL v1 had better com support.

Downgrading using

```
wsl -l -v
# Confirm the <distro_name>
wsl --set-version <distro-name> 1

```

WSL v2 has a few hacks one is a tcp IP com port

https://github.com/microsoft/WSL/issues/6692

Neonel21, 2024-06-15:
Hiya! A long time has passed since the creation of this issue and I was quite erratic about whether to respond with such delay. Truthfuly, I have forgotten to reply, so I beg your pardon. I have solved the issue on my end. The problem stemmed from my usage of the Ubuntu Virtual Machine, which was unable to recognise the USB port. @cyborgdennett you were correct; however, I had no recurring problems after using a Live USB for ubuntu22.

closed 2024-06-15 by Neonel21