google-coral/coralmicro #67: trouble with bash setup.sh

cyborgdennett, 2023-09-14:
### Description

Hello!

I have trouble with the getting started guide

After doing `bash setup.sh`, I get the output as seen below.

The problem lies in `coralmicro/scripts/requirements.txt` where it tries to install hidapi==0.10.1.

I'm no python expert but the failure is in failing to build wheels for this package.

(running on Ubuntu 22.04 on WSL2)

<details><summary>Details</summary>
<p>

Hit:1 http://security.ubuntu.com/ubuntu jammy-security InRelease
Hit:2 https://repositories.intel.com/graphics/ubuntu jammy InRelease                                                                                                         
Hit:3 https://ppa.launchpadcontent.net/kisak/kisak-mesa/ubuntu jammy InRelease                                                                                               
Hit:4 https://cyberbotics.com/debian binary-amd64/ InRelease                                                                                             
Hit:5 http://archive.ubuntu.com/ubuntu jammy InRelease                                                                         
Hit:6 http://archive.ubuntu.com/ubuntu jammy-updates InRelease                                                          
Hit:7 http://archive.ubuntu.com/ubuntu jammy-backports InRelease                                                        
Hit:8 http://packages.ros.org/ros2/ubuntu jammy InRelease           
Hit:9 https://apt.kitware.com/ubuntu jammy InRelease
Reading package lists... Done
W: https://cyberbotics.com/debian/binary-amd64/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
make is already the newest version (4.3-4.1build1).
libhidapi-hidraw0 is already the newest version (0.11.2-1).
libudev-dev is already the newest version (249.11-0ubuntu3.9).
libusb-1.0-0-dev is already the newest version (2:1.0.25-1ubuntu2).
python3-dev is already the newest version (3.10.6-1~22.04).
python3-pip is already the newest version (22.0.2+dfsg-1ubuntu0.3).
cmake is already the newest version (3.27.4-0kitware1ubuntu22.04.1).
The following packages were automatically installed and are no longer required:
  libcups2-dev libcupsfilters-dev libcupsfilters1 libcupsimage2 libcupsimage2-dev libfox-1.6-0 libfox-1.6-dev libtiff5-dev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 406 not upgraded.
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in /home/casper/.local/lib/python3.10/site-packages (23.2.1)
DEPRECATION: distro-info 1.1build1 has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of distro-info or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
DEPRECATION: gpg 1.16.0-unknown has a non-standard version number. pip 23.3 will enforce this behaviour change. A possible replacement is to upgrade to a newer version of gpg or contact the author to suggest that they release a version with a conforming version number. Discussion can be found at https://github.com/pypa/pip/issues/12063
Defaulting to user installation because normal site-packages is not writeable
Collecting hexformat==0.2 (from -r /home/casper/ml/coralmicro/scripts/requirements.txt (line 1))
  Using cached hexformat-0.2-py3-none-any.whl
Collecting hidapi==0.10.1 (from -r /home/casper/ml/coralmicro/scripts/requirements.txt (line 2))
  Using cached hidapi-0.10.1.tar.gz (60 kB)
  Preparing metadata (setup.py) ... done
Collecting progress==1.5 (from -r /home/casper/ml/coralmicro/scripts/requirements.txt (line 3))
  Using cached progress-1.5-py3-none-any.whl
Collecting pyserial==3.5 (from -r /home/casper/ml/coralmicro/scripts/requirements.txt (line 4))
  Using cached pyserial-3.5-py2.py3-none-any.whl (90 kB)
Collecting pyusb==1.2.0 (from -r /home/casper/ml/coralmicro/scripts/requirements.txt (line 5))
  Using cached pyusb-1.2.0-py3-none-any.whl (58 kB)
Requirement already satisfied: setuptools>=19.0 in /home/casper/.local/lib/python3.10/site-packages (from hidapi==0.10.1->-r /home/casper/ml/coralmicro/scripts/requirements.txt (line 2)) (58.2.0)
Building wheels for collected packages: hidapi
  Building wheel for hidapi (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [8 lines of output]
      running bdist_wheel
      running build
      running build_ext
      Compiling hid.pyx because it changed.
      [1/1] Cythonizing hid.pyx
      /tmp/pip-install-b9fr5uzm/hidapi_33bf3e5059c54814b131fec0ca269602/.eggs/Cython-3.0.2-py3.10-linux-x86_64.egg/Cython/Compiler/Main.py:384: FutureWarning: Cython directive 'language_level' not set, using '3str' for now (Py3). This has changed from earlier releases! File: /tmp/pip-install-b9fr5uzm/hidapi_33bf3e5059c54814b131fec0ca269602/hid.pyx
        tree = Parsing.p_module(s, pxd, full_module_name)
      error: unknown file type '.pxd' (from 'chid.pxd')
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for hidapi
  Running setup.py clean for hidapi
Failed to build hidapi
ERROR: Could not build wheels for hidapi, which is required to install pyproject.toml-based projects

</p>
</details> 

### Issue Type

Bug, Build/Install

### Operating System

Windows 10

### Coral Device

Dev Board Micro

cyborgdennett, 2023-09-14:
Okay, so I found that using a newer version of hidapi fixes this issue, I just set the version in requirements.txt to `0.11.2` and the problem was resolved.

cyborgdennett, 2023-09-14:
https://github.com/google-coral/coralmicro/pull/68

hpssjellis, 2023-09-14:
Thanks @cyborgdennett that worked. I do not understand the next instructions. Can anyone explain them or give the specific steps say to run the blink_led code.

```
Generate the project Makefile (run this from the out-of-tree-sample root):

# -B specifies the path for your build output path and
# -S specifies the path to the CMakeLists.txt file.
cmake -B out -S .
Build the app:

make -C out -j4
To maximize your CPU usage, replace -j4 with either -j$(nproc) on Linux or -j$(sysctl -n hw.ncpu) on Mac.
```

I am fine with flashing code, just not used to cmake. I typically use the Arduino IDE.

peepo, 2023-09-17:
upvote:  set the version in requirements.txt to 0.11.2

closed 2023-10-03 by cyborgdennett