google-coral/coralmicro #70: How to flash code on Micro with WSL2?

cyborgdennett, 2023-09-14:
### Description

Hello, I have been trying to get the Micro flashed from WSL.

I have not succeeded.

I posted some comments on another issue I thought would be related to my problem, but maybe that wasn't the case fully.
See below for further explanation an what I tried to solve the issue on my own.

Current status is: the `flashtool.py` is calling `third_party/nxp/blhost` binary with the file to flash to the controller. But this program says `Error: UsbHidPeripheral() cannot open USB HID device (vid=0x1fc9, pid=0x013d, sn=).`

I have used `ubsipd` to forward the usb ports to WSL.

Any help would be highly appreciated.

-------

              Hi, I have the same issue on Windows (WSL) ubuntu22 

```bash
coralmicro$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 18d1:9308 Google Inc. Coral Dev Board Micro
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

coralmicro$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
        Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x00017854 | crc=0xdd594538 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x00017854 | crc=0x9b88b39a | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
Encountered an error during flashing: Unable to find device in SDP mode.
```

After the error, the Coral Micro is not anylonger found is `lsusb`, so I think during the flashing the usb connection stops or something

_Originally posted by @cyborgdennett in https://github.com/google-coral/coralmicro/issues/44#issuecomment-1719281954_
            
              So one solution I found is to spam in powershell:
```powershell
usbipd wsl attach --busid 1-2
```
Then it will go through this step.

next error is as follow:
```bash
coralmicro$ python3 scripts/flashtool.py -e detect_faces
Finding all necessary files
Creating Filesystem
        Section: 0x0
iMX bootable image generated successfully
Boot Section 0x00000000:
  FILL | adr=0x00010000 | len=0x00000004 | ptn=0xc2000105
  FILL | adr=0x00010004 | len=0x00000004 | ptn=0x00010020
  FILL | adr=0x00010008 | len=0x00000004 | ptn=0x00040004
  FILL | adr=0x0001000c | len=0x00000004 | ptn=0x00080004
  FILL | adr=0x00010020 | len=0x00000004 | ptn=0xc0010021
  ENA  | adr=0x00010000 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000004 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x00000008 | cnt=0x00000004 | flg=0x0110
  ERAS | adr=0x0000000c | cnt=0x00000040 | flg=0x0110
  LOAD | adr=0x00000004 | len=0x00017854 | crc=0x93bf7879 | flg=0x0110
  LOAD | adr=0x00000008 | len=0x00017854 | crc=0xf5171da3 | flg=0x0110
STATE_CHECK_FOR_ANY
STATE_CHECK_FOR_CORAL_MICRO
STATE_RESET_TO_SDP
STATE_CHECK_FOR_SDP
STATE_LOAD_FLASHLOADER
Traceback (most recent call last):
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 1193, in <module>
    main()
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 1182, in main
    RunFlashtool(StateCheckForAny, sbfile_path=sbfile_path,
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 821, in RunFlashtool
    (prev_state, state) = (state, state(**state_kwargs))
  File "/home/casper/ml/coralmicro/scripts/flashtool.py", line 543, in StateLoadFlashloader
    subprocess.check_call([blhost_path, '-u', sdp_vidpid(), '--', 'load-image',
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/home/casper/ml/coralmicro/third_party/nxp/blhost/bin/linux/amd64/blhost', '-u', '0x1fc9,0x13d', '--', 'load-image', '/tmp/tmpudxc1pku/ivt_flashloader.bin']' returned non-zero exit status 1.
```

_Originally posted by @cyborgdennett in https://github.com/google-coral/coralmicro/issues/44#issuecomment-1719302239_
            
              You can use 
```powershell
usbipd wsl attach --busid 1-2 -a
```
The -a will make the usbipd service to keep looking for the busid even after it is disonnected for a short while.

I changed line 543 of flashtool.py to: subprocess.DEVNELL->sys.stderr
```python
def StateLoadFlashloader(blhost_path, flashloader_path):
  subprocess.check_call([blhost_path, '-u', sdp_vidpid(), '--', 'load-image',
                        flashloader_path], stdout=sys.stderr, stderr=subprocess.DEVNULL)
  return StateCheckForFlashloader
```
to see output:
```bash
Error: UsbHidPeripheral() cannot open USB HID device (vid=0x1fc9, pid=0x013d, sn=).
```

checking usbipd wsl list in powershell, I get the following two addresses:
18d1:9308 Coral
1fc9:013d USB device(when connecting while holding button)

I guess the problem is that `usbipd` only can attach a single vidpid.

My version of usbipd is 2.4.1, I see the latest is 3.2.0, let me update and check if that works

Edit: 
just updated to 3.2.0 and same issue.
Looking at the output of usbipd I get
```ps
PS C:\WINDOWS\system32> usbipd wsl attach --busid 1-2 -a
usbipd: info: Using default WSL distribution 'Ubuntu-22.04'; specify the '--distribution' option to select a different one.
usbipd: warning: USB filter 'USBPcap' is known to be incompatible with this software; 'bind --force' will be required.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
Attached
Detached
Attached
Detached
usbip: error: Attach Request for 1-2 failed - Device busy (exported)
Attached
```
So as you can see, it attaches and detaches, but when the usb gets an new `vidpid` address, the tool is not made to work with that. I guess what I can do is make a ticket for the usbipd repo

_Originally posted by @cyborgdennett in https://github.com/google-coral/coralmicro/issues/44#issuecomment-1719412210_
            

### Issue Type

Documentation Feature Request

### Operating System

Windows 10

### Coral Device

Dev Board Micro

hpssjellis, 2023-09-20:
@cyborgdennett 

I wonder if their is a web based flash platform like the esp32 people have   https://espressif.github.io/esptool-js/

Perhaps [platformio](https://platformio.org/) could do it.

I can build the files and in the BUILD folder I can see the files including a .hex file. Perhaps we can use windows to flash the .hex file to the coralMicro. I can't remember if the Arduino IDE flashes the .hex or .elf files. We could use a debugger. Anyone got ideas.

cyborgdennett, 2023-09-21:
@hpssjellis 

I found the flashtool of coral arduino. Its located at `C:\Users\<user>\AppData\Local\Arduino15\packages\coral\tools\flashtool\1.0.1`

<details><summary>Details</summary>
<p>

```bash
usage: flashtool.exe [-h] [--build_dir BUILD_DIR] [--elf_path ELF_PATH]
                     [--app APP] [--example EXAMPLE] [--subapp SUBAPP] [--ram]
                     [--noreset] [--serial SERIAL] [--list] [--ums]
                     [--usb_ip_address USB_IP_ADDRESS]
                     [--dns_server DNS_SERVER]
                     [--ethernet_config ETHERNET_CONFIG]
                     [--wifi_ssid WIFI_SSID] [--wifi_psk WIFI_PSK]
                     [--wifi_config WIFI_CONFIG] [--ethernet_speed {10,100}]
                     [--debug] [--jlink_path JLINK_PATH] [--cached]
                     [--noprogram] [--nodata]

Coral Dev Board Micro flashtool

optional arguments:
  -h, --help            show this help message and exit

Flashing options:
  --build_dir BUILD_DIR, -b BUILD_DIR
                        Path to the coralmicro build output. (default:
                        C:\Users\caspe\AppData\Local\Temp\_MEI120522\..\build)
  --elf_path ELF_PATH   Path to the .stripped binary (ELF file) to flash. This
                        must be the full path and filename. Usually used for
                        out-of-tree projects only. (default: None)
  --app APP, -a APP     Name of the coralmicro in-tree "app" to flash. This
                        must be a project directory name available in
                        build/apps/. (default: None)
  --example EXAMPLE, -e EXAMPLE
                        Name of the coralmicro in-tree "example" to flash.
                        This must be a project directory name available in
                        build/examples/. (default: None)
  --subapp SUBAPP       The target name you want to flash. This is needed only
                        when using --app or --example and the target name is
                        different than the project directory name given to
                        those arguments. (default: None)
  --ram                 Flashes the app to the RAM only, instead of to flash
                        memory. This means the app will be overrwritten upon
                        reset. Without this flag, it flashes to the flash
                        memory and resets the board to load it. (default:
                        False)
  --noreset             Prevents resetting the board after flashing it.
                        (default: False)
  --serial SERIAL       The board serial number you want to flash. This is
                        necessary only if you have multiple Coral Dev Board
                        Micro devices connected. (default: None)
  --list                Prints all detected Coral Dev Board Micro devices.
                        (Does not flash the board.) (default: False)
  --ums                 Puts the device in a state where it acts as an LFS
                        mountable USB drive.
                        (Does not flash the board.) (default: False)

Board network settings:
  --usb_ip_address USB_IP_ADDRESS
                        The board IP address for Ethernet-over-USB
                        connections. (default: 10.10.10.1)
  --dns_server DNS_SERVER
                        The DNS server to use for network address resolution.
                        By default, the board uses the DNS specified by the
                        DHCP server, so this should be specified if using a
                        static IP with --ethernet_config. (default: None)
  --ethernet_config ETHERNET_CONFIG
                        Path to a text file that specifies Ethernet network IP
                        settings to use on the board (if you are not using a
[... 28 lines cut ...]
Debugging options:
  --debug               Loads the app into RAM, starts the JLink debug server,
                        and attaches GDB. You must have a JTAG debugger
                        attached to the board, which requires an add-on board
                        with the JTAG pins exposed, such as the Coral Wireless
                        Add-on. For details, see www.coral.ai/docs/dev-board-
                        micro/wireless-addon/#jtag. (default: False)
  --jlink_path JLINK_PATH
                        Path to JLink if --debug is enabled. (default:
                        /opt/SEGGER/JLink)

Advanced options:
  --cached              Flashes using cached target files, which can speed up
                        the flashing process. You must first run
                        scripts/cache_build.py. (default: False)
  --noprogram           Prevents flashing the program code (only data is
                        flashed, unless --nodata is also specified). (default:
                        False)
  --nodata              Prevents flashing the app data (only program code is
                        flashed, unless --noprogram is also specified).
                        (default: False)
```

</p>
</details>

hpssjellis, 2023-09-25:
> C:\Users\<user>\AppData\Local\Arduino15\packages\coral\tools\flashtool\1.0.1

Very cool, have you managed to flash the code using the arduino IDE? I will test  using the Ubuntu link from windows and also copying the code to the same folder.

hpssjellis, 2023-09-26:
@cyborgdennett what about compiling the coralMicro on WSL but for Arduino. They both use the gcc compiler so it should be able to be done, then when using the Arduino IDE the exported files should be in the correct .elf format.

ansemjo, 2024-04-11:
I've hit this same issue as well last week and found a solution. Basically, at the moment you need to compile and use your own kernel for WSL2; one which supports `CONFIG_USB_HIDDEV` and `CONFIG_HIDRAW`. Additionally, I found even `usbipd ... --auto-attach` to be too slow and used the [WSL USB GUI](https://gitlab.com/alelec/wsl-usb-gui) instead. Otherwise the `flashtool.py` would appear to time-out waiting for the Micro to reappear. The rest here is the guide that I wrote for my students:

---
According to [WSL Issue #8302](https://github.com/microsoft/WSL/issues/8302), the kernel config options `CONFIG_USB_HIDDEV` and `CONFIG_HIDRAW` are now enabled by default in the `Microsoft/config-wsl` configuration. These options are needed to properly attach forwarded USB devices in WSL2 and avoid the following error in the kernel log:

````
hid-generic 0003:1FC9:013D.0011: device has no listeners, quitting
````

In my fully-updated WSL2 installation of Debian (on 2024-04-04) these options were not enabled; you can check that with `zgrep -E "_HID(DEV|RAW)" /proc/config.gz`. The WSL2 kernel is provided by Windows and should be distributed to your system at some time in the future. Unfortunately, this means that right now you'll have to compile your own kernel with the options enabled. Let's try to follow the following guide by Microsoft – just not with the v6 kernel but the version 5.15.150.1 linked in the issue above.

> https://learn.microsoft.com/en-us/community/content/wsl-user-msft-kernel-v6

* Enter a WSL terminal; you'll want to compile the kernel inside a Linux environment.
* Download the release [linux-msft-wsl-5.15.150.1.tar.gz](https://github.com/microsoft/WSL2-Linux-Kernel/archive/refs/tags/linux-msft-wsl-5.15.150.1.tar.gz) and unpack the archive. Then enter the directory: <br>
  `cd WSL2-Linux-Kernel-linux-msft-wsl-5.15.150.1/`
* Install necessary build dependencies: <br>
  `sudo apt update && sudo apt install build-essential flex bison libssl-dev libelf-dev bc python3 pahole rsync`
* Build the kernel with the default config: <br>
  `make -j$(nproc) KCONFIG_CONFIG=Microsoft/config-wsl`
* Install headers and modules inside your WSL distribution: <br>
  `sudo make modules_install headers_install`
* Copy the Kernel into Windows' filesystem: <br>
  `cp arch/x86/boot/bzImage /mnt/c/Users/<username>/bzImage-5.15.150.1`
* Exit the WSL terminal and shutdown the instance. Then configure the new kernel following the steps in section 2 of the guide linked above.

When you start a new WSL2 terminal now, you should see the new kernel being used:

````
$ uname -r
5.15.150.1-microsoft-standard-WSL2
$ zgrep -E "_HID(DEV|RAW)" /proc/config.gz
CONFIG_HIDRAW=y
CONFIG_USB_HIDDEV=y
````

Now follow the steps from the [`usbipd` Guide](https://learn.microsoft.com/windows/wsl/connect-usb) by Microsoft and make sure that you can attach USB devices like the Coral Micro _in general_. You can check successful attachment from inside WSL with `lsusb`.

However, the manual approach with `usbipd` only partially works because during flashing the Coral Micro is reset multiple times and attaches using different USB vendor and product IDs. Even using the `--auto-attach` flag does not appear to reattach devices fast enough and the `flashtool.py` times out waiting for the device. To work around this, there's a [WSL USB GUI](https://gitlab.com/alelec/wsl-usb-gui), which seems to attach devices a bit faster.

* Download and install a recent release; at the time of writing [WSL-USB-5.3.0.msi](https://gitlab.com/api/v4/projects/35133362/packages/generic/wsl-usb-gui/5.3.0/WSL-USB-5.3.0.msi) is the latest.
* Start the WSL USB Manager GUI and find the Coral Dev Board Micro device (in the Screenshots it's on bus ID `2-2`).
* Check the `forced` box for the device. Select the line again and click on the `Auto-Attach` button in the "Forwarded Devices" panel. In the dialog, pick **Port** to always automatically attach any new USB devices _at this same USB port_. When you use a different USB port, you may need to repeat this configuration.

cyborgdennett, 2024-04-12:
I have this in .wslconfig: 

[wsl2]
kernel=C:\\Users\\caspe\\bzImage-5.15.150.1

But when I `wsl --shutdown` and check the version it is still

uname -r
5.15.133.1-microsoft-standard-WSL2