google-coral/coralmicro #72: Amazing product the CoralMicro, but where is the support?

hpssjellis, 2023-09-26:
### Description

**Amazing product the CoralMicro, but where is the support?**

As far as I can tell this amazing product has almost zero upkeep and near zero support. I don't think the [Tensorflow Forum](https://discuss.tensorflow.org/) or [Stackoverflow](https://stackoverflow.com/questions/tagged/google-coral) are answering questions and I don't think Coral.ai has setup a forum. 

The main support page is https://coral.ai/support/ where we email information and hope they get back to us.

Can people list others interested in this board so that we can have some kind of a community.

@mbrooksx   @cyborgdennett @arthurkafer  @teamrevspace  @atvrager  @dmitriykovalev  @josecols @scottamain  @r-clifford  @having11 

Does any users of this board have any suggestions other than just to keep posting issues?

For me I have lots of questions about the Arduino support, however am finding things running fairly smoot on Linux Ubuntu.

peepo, 2023-10-03:
the most obvious immediate benefit might be if people created an issue here to link to their github repositories, when they have something to show? seems awkward, and only addresses a particular social need, but...
as yet I've nothing to show: https://github.com/peepo/

hpssjellis, 2023-10-04:
@peepo I don't have much either. Here is the link I would put things at   

[https://github.com/hpssjellis/mcu-stable-edu-launch/tree/main/google-coral-micro](https://github.com/hpssjellis/mcu-stable-edu-launch/tree/main/google-coral-micro
)

Basically most things work on full Ubuntu Linux, but the Arduino parts are flakey at best. If I could get things working I would probably show a ton of Arduino examples using the CoralMicro.

OctavioFM, 2024-03-22:
I got one two days ago. The built-in face-detection program worked well, but after trying to flash it with Arduino IDE, the board became unresponsive. Haven't been able to make it work after that.

peepo, 2024-03-22:
Did anyone find a commercial purpose for the CoralMicro?
Google does have it's limitations....

arthurkafer, 2024-03-25:
> I got one two days ago. The built-in face-detection program worked well, but after trying to flash it with Arduino IDE, the board became unresponsive. Haven't been able to make it work after that.

There were some problems I've experienced using Arduino IDE in Windows, in cases like yours. 
I changed to Linux and flashing code directly by CLI and it ran again normally

arthurkafer, 2024-03-25:
> Did anyone find a commercial purpose for the CoralMicro? Google does have it's limitations....

Yes actually, but where I work we're in board developing phase, nothing physically running anything right now. It is easier to find other microprocessors and develop something that already has support, because of time limitations