google-coral/coralmicro #73: Coral Dev Board Micro USB is not recognized

dadanugm, 2023-09-30:
### Description

Hi,

Would like to seek help for an issue with Coral Dev Board Micro. I am using Arduino IDE to flashing Coral Micro, but suddenly after flashing the USB board is not recognized anymore. I tried to solve it by following this https://coral.ai/docs/dev-board-micro/get-started#serial-downloader, but this solution is not working. Basically I perform Serial Downloader mode but the orange led is never turn on.
any Idea to solve this issue?

[image]

Thanks

### Issue Type

Build/Install

### Operating System

Windows 10

### Coral Device

Dev Board Micro

### Programming Language

C++

bogdannedelcu, 2023-10-02:
I had similar problem on Win an Linux, I replaced with a thicker and shorter usb cable and is working now

dadanugm, 2023-10-14:
Hi @bogdannedelcu 
Thanks for your response, may I know the usb cable brand that you use?. I tried multiple cable but not working

bogdannedelcu, 2023-10-14:
No brand name written. It came with a SanDisk. As you may see it is thicker
than the other one which did not work.