google-coral/coralmicro #77: Forum

peepo, 2023-10-03:
Apologies, not one for stackoverflow....

Many popular boards like the Raspberry Pi have a developer forum. 
Surely the Google Coral boards need a developer forum, 
and in particular the coralmicro. Where is it?

ie code is not the only thing to discuss, sharing projects etc.

hpssjellis, 2023-10-03:
@peepo I agree. This impressive shiny new board does not have support anywhere, almost like Google has already deprecated it. I love the board but will probably have to go back to the Esp32S3 or Arduino products unless the CoralMicro starts looking like it has some future. 

[Amazing product the CoralMicro, but where is the support? #72](https://github.com/google-coral/coralmicro/issues/72)

dadanugm, 2023-10-14:
@hpssjellis Couldn't agree more, feels like soon this board will be abandoned. I was so happy when I got chance to explore this board. but due to this boards unknown issue, now I think I will still stick to sipeed products for some times.