google-coral/coralmicro #102: Compilation fails with missing file in Arduino IDE 2.3.2 with MacOS Sonoma (14.2.1)

binloan, 2024-03-18:
### Description

Following this guide for setup:
[https://coral.ai/docs/dev-board-micro/arduino/](https://coral.ai/docs/dev-board-micro/arduino/)
After successful install even when trying to compile an empty sketch the following issue appears:
```
/Users/XXX/Library/Arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino/include/libs/camera/camera.cc:27:10: fatal error: third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h: No such file or directory
   27 | #include "third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

I got the same issue when I tried compiling on Ubuntu 22.04. I'm using Ardunio IDE 2.3.2.
Interestingly it works without issues on Windows 10 (even though it's not supported based on documentation).
Does anyone have a similar issue? Thank you.

### Issue Type

Bug

### Operating System

Mac OS

### Coral Device

Dev Board Micro

### Programming Language

C++

### Relevant Log Output

```shell
/Users/XXX/Library/Arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino/include/libs/camera/camera.cc:27:10: fatal error: third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h: No such file or directory
   27 | #include "third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

jakubcizek, 2024-04-06:
Got same error on Ubuntu and Windows 11.  Any workaround after three weeks?

binloan, 2024-04-08:
> Got same error on Ubuntu and Windows 11. Any workaround after three weeks?

No. It seems the sources are just broken. I tried on MacOS too.

djolodjolo23, 2024-04-11:
Same problem here, on Ubuntu.

jamstooks, 2024-04-22:
Hope we can fix or find a workaround as there are some really handy examples with Arduino dependencies.

0xCharlottez, 2024-05-06:
The [third_party](https://github.com/google-coral/coralmicro-rt1176-sdk/tree/af651301e1013c6722fd6fd36aff0cdc73d5b575/devices/MIMXRT1176) folder under the main github page contains the missing file. So I manually copied the third_party folder to my own arduino folder and this error can be resolved. There are a series of other files need manual copy paste from the main branch, and eventually I left with an error like this: 
`[sdads](arm-none-eabi-gcc: error: /home/User/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cflags.txt: No such file or directory)`

hengj10, 2024-05-23:
Same problem here, have you solved this problem?> The [third_party](https://github.com/google-coral/coralmicro-rt1176-sdk/tree/af651301e1013c6722fd6fd36aff0cdc73d5b575/devices/MIMXRT1176) folder under the main github page contains the missing file. So I manually copied the third_party folder to my own arduino folder and this error can be resolved. There are a series of other files need manual copy paste from the main branch, and eventually I left with an error like this: 
> 
> `[sdads](arm-none-eabi-gcc: error: /home/User/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cflags.txt: No such file or directory)`

hengj10, 2024-05-23:
Same problem here, have you solved this problem?

hengj10, 2024-05-23:
> The [third_party](https://github.com/google-coral/coralmicro-rt1176-sdk/tree/af651301e1013c6722fd6fd36aff0cdc73d5b575/devices/MIMXRT1176) folder under the main github page contains the missing file. [...]

Same problem here, have you solved this problem?

vmanoj1996, 2024-07-13:
I am exactly in this same spot. Unable to find that cflags.txt file.
Anyone figured it yet?

Hello9999901, 2024-07-13:
TLDR; tested macOS solution: use Arduino IDE 2.0.0 (confirmed) or similarly old build

Hello everyone, I found a workaround to this problem by using an older version of Arduino IDE. In my case, my system is:
macOS Sonoma 14.5 on an Apple M3 Max 14" running Rosetta emulation of Arduino IDE 2.0.0. Everything is standard. I ran into the same issues as @0xCharlottez and @binloan. Just tried the blinkLED example and it works as expected.

My suspicion is that arm64 Arduino IDE builds are doing something breaking. I will dig into this further if I have time.

Hello9999901, 2024-07-19:
Update: I unfortunately won't look into this much further. Our team dropped the coral in favor of the Teensy 4.0/4.1.