google-coral/coralmicro #115: Issue building project on Rasberry Pi 5

antypko, 2024-04-21:
### Description

Following this guide for building an example:
https://coral.ai/docs/dev-board-micro/get-started/#5-try-the-face-detection-example

During building process I am getting error:

`CMake Error at /usr/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
  The C compiler

    "/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_dd72c/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_dd72c.dir/build.make CMakeFiles/cmTC_dd72c.dir/build
    gmake[1]: Entering directory '/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL'
    Building C object CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc   -Wall -Wno-psabi -mthumb -fno-common -ffunction-sections -fdata-sections -ffreestanding -fno-builtin -mapcs-frame --specs=nano.specs --specs=nosys.specs -u _printf_float -ffile-prefix-map=/home/sviatoslavchetyrkin/coral/coralmicro= -std=gnu99  -o CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj -c /home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL/testCCompiler.c
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 1: ELF: not found
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 2: Syntax error: "(" unexpected
    gmake[1]: *** [CMakeFiles/cmTC_dd72c.dir/build.make:78: CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj] Error 2
    gmake[1]: Leaving directory '/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL'
    gmake: *** [Makefile:127: cmTC_dd72c/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:22 (project)`

I am using Raspberry Pi 5 as development station and have installed python 3.11.2

### Issue Type

Build/Install

### Operating System

Linux

### Coral Device

Dev Board Micro

### Programming Language

Other

### Relevant Log Output

```shell
+ cmake -B /home/sviatoslavchetyrkin/coral/coralmicro/build -G 'Unix Makefiles' -S /home/sviatoslavchetyrkin/coral/coralmicro
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 1: ELF: not found
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 2: Syntax error: "(" unexpected
-- Toolchain prefix: /
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Detecting C compiler ABI info
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 1: ELF: not found
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 2: Syntax error: "(" unexpected
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 1: ELF: not found
/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 2: Syntax error: "(" unexpected
-- Check for working C compiler: /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
  The C compiler

    "/home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_dd72c/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_dd72c.dir/build.make CMakeFiles/cmTC_dd72c.dir/build
    gmake[1]: Entering directory '/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL'
    Building C object CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc   -Wall -Wno-psabi -mthumb -fno-common -ffunction-sections -fdata-sections -ffreestanding -fno-builtin -mapcs-frame --specs=nano.specs --specs=nosys.specs -u _printf_float -ffile-prefix-map=/home/sviatoslavchetyrkin/coral/coralmicro= -std=gnu99  -o CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj -c /home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL/testCCompiler.c
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 1: ELF: not found
    /home/sviatoslavchetyrkin/coral/coralmicro/third_party/toolchain-linux/gcc-arm-none-eabi-9-2020-q2-update/bin/arm-none-eabi-gcc: 2: Syntax error: "(" unexpected
    gmake[1]: *** [CMakeFiles/cmTC_dd72c.dir/build.make:78: CMakeFiles/cmTC_dd72c.dir/testCCompiler.c.obj] Error 2
    gmake[1]: Leaving directory '/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeScratch/TryCompile-ImpHZL'
    gmake: *** [Makefile:127: cmTC_dd72c/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:22 (project)

-- Configuring incomplete, errors occurred!
See also "/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeOutput.log".
See also "/home/sviatoslavchetyrkin/coral/coralmicro/build/CMakeFiles/CMakeError.log".
```

andrewtvuong, 2024-06-30:
I have this same issue, did you ever resolve this?

antypko, 2024-07-09:
No, I didn't. In my case CoralMicro SDK does not have support for ARM chips as dev env.

andrewtvuong, 2024-07-09:
I kind of got it working to detect the coral micro but gave up for now..  too much time wasted

https://github.com/google-coral/coralmicro/issues/120

stuck on this part:

(venv) ➜  coralmicro git:(main) ✗ python3 scripts/flashtool.py -a multicore_model_cascade        

/tmp/tmp_bcg2yq1/ivt_flashloader.bin does not exist