google-coral/coralmicro #121: Arduino Demo - DetectFaces does not compile - unable to find -  ```cpp fsl_cache.h ```

vmanoj1996, 2024-07-13:
### Description

The supplied demo - DetectFaces - for Arduino does not compile due to the following error,

```
/home/<user>/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino/include/libs/camera/camera.cc:27:10: fatal error: third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h: No such file or directory
   27 | #include "third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

`cm7` folder was missing completely.

 I only see the following files under "...third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers",

`
fsl_clock.h	  fsl_csi.h    fsl_lpi2c_freertos.h  fsl_lpspi.h     fsl_pwm.h
fsl_common_arm.h  fsl_edma.h   fsl_lpi2c.h	     fsl_pdm_edma.h  fsl_pxp.h
fsl_common.h	  fsl_lpadc.h  fsl_lpspi_freertos.h  fsl_pdm.h	     fsl_sema4.h
`

Were these examples tested before?

### Issue Type

Bug

### Operating System

Ubuntu

### Coral Device

Dev Board Micro

### Programming Language

C++

### Relevant Log Output

```shell
FQBN: coral:coral_micro:coral_micro
Using board 'coral_micro' from platform in folder: /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1
Using core 'arduino' from platform in folder: /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1

Detecting libraries used...
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/includes.txt -o /dev/null /tmp/arduino/sketches/295A16C340807D9DA50CE5C2ECDB74F5/sketch/DetectFaces.ino.cpp
Alternatives for coralmicro_SD.h: [CoralMicro_SD@1.0.0]
ResolveLibrary(coralmicro_SD.h)
  -> candidates: [CoralMicro_SD@1.0.0]
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/includes.txt -o /dev/null /tmp/arduino/sketches/295A16C340807D9DA50CE5C2ECDB74F5/sketch/DetectFaces.ino.cpp
Alternatives for coralmicro_camera.h: [CoralMicro_Camera@1.0.0]
ResolveLibrary(coralmicro_camera.h)
  -> candidates: [CoralMicro_Camera@1.0.0]
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_Camera/src -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/includes.txt -o /dev/null /tmp/arduino/sketches/295A16C340807D9DA50CE5C2ECDB74F5/sketch/DetectFaces.ino.cpp
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_Camera/src -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/includes.txt -o /dev/null /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src/File.cpp
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_Camera/src -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/includes.txt -o /dev/null /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src/coralmicro_SD.cpp
/home/manoj/.arduino15/packages/coral/tools/arm-none-eabi-gcc/9-2020q2/bin/arm-none-eabi-g++ -c -w -Os -g -nostdlib @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/defines.txt @/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro/cxxflags.txt -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -w -x c++ -E -CC -DARDUINO=10607 -DARDUINO_coral_micro_FREERTOS -DARDUINO_ARCH_CORAL_MICRO -DARDUINO_LIBRARY_DISCOVERY_PHASE=1 -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/cores/arduino -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/variants/coral_micro -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD/src -I/home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_Camera/src -iprefix/home/manoj/.arduino15/packages/coral/hardware/coral_micro [... cut ...]
   27 | #include "third_party/nxp/rt1176-sdk/devices/MIMXRT1176/drivers/cm7/fsl_cache.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

Using library CoralMicro_SD at version 1.0.0 in folder: /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_SD 
Using library CoralMicro_Camera at version 1.0.0 in folder: /home/manoj/.arduino15/packages/coral/hardware/coral_micro/1.0.1/libraries/CoralMicro_Camera 
exit status 1

Compilation error: exit status 1
```

ajayIhub, 2024-07-22:
I am also facing this issues with arduino ide

wawanUnic, 2024-08-13:
I am also facing this issues with arduino ide

tommiesatellite, 2024-12-25:
I am also having this issue with Arduino die 2.3.3

vmanoj1996, 2024-12-26:
The FreeRTOS examples run fine though.

tommiesatellite, 2024-12-26:
I also have the same issue with FreeRTOS

vmanoj1996, 2024-12-26:
Might be a mac issue. I did it on Ubuntu 22.04.

tommiesatellite, 2024-12-26:
Thank you, that gives me a lot of hope.

tommiesatellite, 2024-12-26:
It seems that uninstalling my Arduino v2.3.3 and installing v2.0.4 is the trick.  It now works on my MacOS with Arduino v2.0.4.  That older release can be downloaded at: https://github.com/arduino/arduino-ide/releases/tag/2.0.4